Koenkk/zigbee2mqtt #1814: Cannot get the Node Descriptor Error

wes1993, 2019-08-05:
Hello, i have the same issue, how can i solve the problem?
i'm trying to add the Xiaomi DJT11LM sensor but i can't (zigbee2mqtt:error 8/5/2019, 9:59:48 AM Cannot get the Node Descriptor of the Device: 0x00158d0002af97b2 (Error: Timed out after 10000 ms)) i don't have any string of this device in the database.
i have also 2 ikea tradfri smart switch (plug) that i think work like a router.
The fw of my coordinator is the latest 3.0
My HA Environment is the latest.
My zigbee2MQTT version is: 1.5.1.1
Thanks a lot for all your help.

Koenkk, 2019-08-05:
Duplicate of #1779 

Fix: https://github.com/Koenkk/zigbee2mqtt/issues/1779#issuecomment-515774316

Note that the issue tracker is meant for tracking bugs, for questions please use http://zigbee2mqtt.discourse.group/ or https://discord.gg/dadfWYE.

closed 2019-08-05 by Koenkk

wes1993, 2019-08-05:
Sorry!!! thanks a lot!! :-D