Koenkk/zigbee2mqtt #1815: Konke Pro-Key multi function button

elpaul, 2019-08-05:
Hi everyone.
I have several Xiaomi devices and several Osram devices. They all work without a problem and they also pair up smoothly.
Today I got the KONKE button. Konke Pro-Key multi function button:
[image]
Unfortunately, when I try to pair, I do not receive anything in the logs.
Where to look for a problem?

zigbee2mqtt version: 1.5.1
CC253X firmware version: CC2531_DEFAULT_20190608.

ryanbeaton, 2019-08-05:
https://github.com/Koenkk/zigbee2mqtt/issues/1689#issuecomment-517987242

elpaul, 2019-08-05:
I am asking for understanding, I am fresh on the subject.

> it's work only on 15, 20, 25 channels

What does it mean?

ryanbeaton, 2019-08-05:
Cool. Hope that works.
Sorry for the short response before just saw your issue come in and wanted to get you some info quickly but only had a sec.

Koenkk, 2019-08-05:
Thanks @ryanbeaton !

closed 2019-08-05 by Koenkk