Koenkk/zigbee2mqtt #1817: "raw" networm map should return more information

carldebilly, 2019-08-05:
# Bug Report
Network Map: there's more info in the _graphviz_ version than the `raw` version.

Using _RAW_, we're getting those informations:
* `ieeeAddr` (hardware id)
* `nwkAddr` (network id)
* `lqi` (link_quality?)
* `parent` (may we have more than one?)
* `status`: `online` / `offline`

Using _Graphviz_, there's more information:
* **Type** (Coordinator/Router/EndDevice
* **Friendly Name**
* **Model** / **Manufacturer**
* Links + quality

## What did you expect to happen
It would be nice to get, at least, the _FriendlyName_, _Model_ and _Manufacturer_ in the _Raw_ version.

## Debug Info
zigbee2mqtt version: `1.4.0`
CC253X firmware version: `20190425`

clockbrain, 2019-08-06:
Much of what you are looking for was added in https://github.com/Koenkk/zigbee2mqtt/pull/1676 which is included in v1.5.1

> A side-effect of these code changes is that the 'raw' network map is much richer having both nodes (devices) and link information

carldebilly, 2019-08-06:
Ooohhh that's nice. Thanks! I'll need to upgrade to latest version :-)

closed 2019-10-12 by stale[bot]