Koenkk/zigbee2mqtt #1818: ptvo.switch

olili, 2019-08-05:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

olili, 2019-08-05:
I'm using the CC253x firmware for DIY switches provided by ptvo.org. 
Really great! Also the support of Zigbee2MQtt seems to work almost perfect with my ESP18-MS1 (CC2530+CC2592).  
All outputs and the inputs "bottom_left" and "bottom_right" are working properly.  

But:  
until now I was not able to read in the input 3 and 4, i.e. "top_left" and "top_right".  
I tried to assign different ports to these inputs in order to exclude HW issues, but no progress. 

Any hints? 

O.

Koenkk, 2019-08-06:
@ptvoinfo could you help?

ptvoinfo, 2019-08-07:
@olili Could you please attach a screenshot with your firmware settings. I'll try it with my chip. It would be great if you can post a link to your chip. I cannot find it on Google.

olili, 2019-08-07:
Dear all,

thanks for your reply.

Requested info is available at http://niklas.ml/issue.zip and
http://niklas.ml/E18_User.pdf

I think it is not a matter of HW. Also the control ports for CC2592 are not
used.

Maybe you can support.

Thx in advance.

ptvoinfo, 2019-08-07:
@olili Did you try to delete the device information from the database when you've updated the firmware? Otherwise Z2M may not update information about end points.

olili, 2019-08-07:
I did it once more - stopped z2m, removed db entry, reflashed fw, re-paired -> No change.
Input 3 is still not supported. 

I swapped the port assignments as well. 
CC2530-Ports that are working as  Input1+2, are not not working, as soon those ports were assigned to Input 3+4.

Any hints`?

ptvoinfo, 2019-08-14:
@olili I've fixed this bug in the firmware. Could you please re-download it and check on your side?
https://ptvo.info/wp-content/uploads/2018/09/cc2530_io_1.2.2a.44539_firmware.zip

olili, 2019-08-14:
Dear ptvoinfo, 
it's working now! Once again many thanks for your great work! O.

closed 2019-10-20 by stale[bot]