Koenkk/zigbee2mqtt #1820: [Feature request] Full support Zigbee 3.0 standard: Child Device Management, Parent Announce Command and Install Codes

dzungpv, 2019-08-06:
@Koenkk Now Zigbee HA 1.2 stable but not for all device, you can see my unsolved issue here: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/826930/3058724#3058724
Zigbee 3.0 solve the issue but it is not fully support in zigbee2mqtt.
You also nearly disable it in the https://github.com/Koenkk/Z-Stack-firmware/commit/00a421b26b261f096b064bb993dcc9519b03243b

So some features must add to fix the issue and full compatible with Zigbee 3.0:
- Configurable child aging timeout, first globally and if possible for each devices (router, coordinator), openhab support it globally: https://www.openhab.org/addons/bindings/zigbee/
- If zigbee2mqtt upgrade directly to 3.0 **Parent Announce Command** my not implementation, it is require for  **Child Device Management** works: http://www.ti.com/lit/an/swra615a/swra615a.pdf "The Parent Announce message is now mandatory on all routing devices (router and coordinator) and is used to notify other routing devices in a network of which child devices are known to it. The message is broadcast to 0xFFFC if a routing device previously in the network was rebooted and successfully rejoins the network. It is sent 10 to 20 seconds after this device rejoins the network"
- Install Codes must be add, openhab now support it with setting format like: IEEE Address:Install Code

dzungpv, 2019-08-08:
@Koenkk I have build Router 3.0 from SampleLight and it work fine with CC2530+2591 ZNP 3.0 firmware, no more bug with router like https://github.com/Koenkk/zigbee2mqtt/issues/1316.
Coordinator 3.0 must be rebuilt with child aging enable , in my build i set the timeout to 0 (10 seconds). I am not test this router with any Xiaomi sensor

So to have a stable mesh network with router (TI solution):
1, Use ZNP 3.0.x or 3.x.0
2, All router devices in the network must running Zigbee 3.0 standard
3, End devices (most off sensor) could running any zigbee version.

I ask TI expert and they say we don't need to care about **Parent Announce Command**, it is handle inside ZNP firmware for router and coordinator.

I will share Router zigbee 3.0 here, it work out off the box , this build now only work with CC2530.
Just allow join and then flash the firmware, it will automatically join as DNCKAT_S001
And you can build ZNP firmware with child aging enable for anyone want to test.
[Router_30_cc2530.zip](https://github.com/Koenkk/zigbee2mqtt/files/3480731/Router_30_cc2530.zip)

dzungpv, 2020-04-05:
@Koenkk Please keep this open

dzungpv, 2020-04-05:
@Koenkk I found that current z2m not handle **Parent Announce Command** when starting.
When power ZNP CC2652/1352 for the first time and start z2m right after that, it does send **Parent Announce Command** (it is automatic by z-stack), but if stop z2m and start, it not send, so we must send Parent Announce from z2m every time it start to prevent some bug like this: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/890697

Koenkk, 2020-04-06:
@dzungpv I checked but this method is not exposed on the ZNP api, can you check if it can be added to the startupFromApp? (https://github.com/Koenkk/zigbee-herdsman/blob/25869ccea20fdb2ff1cc76949dbd912b7fbb3558/src/adapter/z-stack/adapter/startZnp.ts#L128)

dzungpv, 2020-04-06:
@Koenkk it automatically call by ZDO:
```
    if ( startMode == MODE_HARD )
    {
      ZDApp_ChangeState( DEV_COORD_STARTING );
      ret = NLME_NetworkFormationRequest( zgConfigPANID, zgApsUseExtendedPANID, runtimeChannel,
                                          zgDefaultStartingScanDuration, beaconOrder,
                                          superframeOrder, false, false, 0 );
    }
    else if ( startMode == MODE_RESUME )
    {
      // Just start the coordinator
      ZDApp_ChangeState( DEV_COORD_STARTING );
      ret = NLME_StartRouterRequest( beaconOrder, beaconOrder, false );
    }
```
zd_app.c
```
  if ( nwkStatus == ZSuccess )
  {
    // Successfully started a ZigBee network
    if ( devState == DEV_COORD_STARTING )
    {
.
.
.
      ZDApp_SetParentAnnceTimer();
```

So i dont see what is the option of  znp.request(Subsystem.ZDO, 'startupFromApp'?

Koenkk, 2020-04-06:
I think startupFromApp is not always called (when it's already running e.g.). But as we cannot call it remotely it should be added somewhere in the firmware.

dzungpv, 2020-06-05:
Unstale

closed 2020-07-26 by stale[bot]