Koenkk/zigbee2mqtt #1821: Eurotronic SPZB0001 get local temperature calibration

airens, 2019-08-06:
## Get method doesn't work as I expect
Hello, I can set Local Temperature Calibration (0x0010) using **zigbee2mqtt/<device_id>/set** topic with payload { "local_temperature_calibration": 2 } (the value of local temperature changing responsively), but when I use **get** method with payload { "local_temperature_calibration": "" }, the parameter appears in the next package from device, but it always zero. How can I get current correct value?

Koenkk, 2019-08-06:
What do you see in the log when executing this command?

airens, 2019-08-07:
> What do you see in the log when executing this command?

```
zigbee2mqtt:info 8/7/2019, 2:20:29 PM Zigbee publish to device 'heat_living_room', hvacThermostat - read - [{"attrId":16}] - {"manufSpec":0,"disDefaultRsp":0} - 1
zigbee2mqtt:error 8/7/2019, 2:20:59 PM Zigbee publish to device 'heat_living_room', hvacThermostat - read - [{"attrId":16}] - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
```

Koenkk, 2019-08-08:
Probably the device is sleeping when you execute the command, does waking it up while doing this solve the problem? (wake it up by pressing a few buttons on it).

airens, 2019-08-09:
After some time, correct values started to come, I don't now what it was, but now it works

closed 2019-08-09 by airens