Koenkk/zigbee2mqtt #1827: cannot pair trust ZYCT-202 remote. Cannot get the Node Descriptor of the Device

s235711, 2019-08-07:
# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: latest
CC253X firmware version: latest

the only relevant log entry is this:
Cannot get the Node Descriptor of the Device: 0x00158d000131435c (Error: Timed out after 10000 ms)

reproduce:
Reset the remote (10 sec press off)
press two bulbs button until all the lights are on

no entry on database nor configuration of 0x00158d000131435c

closed 2019-08-07 by s235711

james-fry, 2019-09-05:
@s235711  I have the same problem.
I had one of these trust controls and managed to pair it.
It stopped working to I tried to re-pair it and it didnt work
Now I got a replacement and I cannot pair that either.

IMO 1.5.1. intruduces issues to pair devices.

Did you manage to get working eventually?

s235711, 2019-09-05:
Nope. I repair it several times and every time it stop working. Im using now tradfri remotes.

james-fry, 2019-09-05:
OK thanks for the info.
I hoped I just got unluckly but looks like the amazon reviews might be justified - even with zigbee2mqtt.
Shame as these controls have (had) a lot of potential...

james-fry, 2019-09-08:
Just updating that I tried a second device and it too had the same problem.
Its a real shame as these are in theory neat little devices, but totally unreliable :(