Koenkk/zigbee2mqtt #1828: Problems with Xiaomi/Honeywell smoke detector

elsaesk, 2019-08-07:
Hi all,

I paired a new Xiaomi/Honeywell smoke detector with success by pressing 3 times the pairing button.

But after the pairing process no data is sent by the detector.

My older Xiaomi/Honeywell smoke detectors work without a problem.

What do I have to do?

Thanks in advance.

elsaesk, 2019-08-12:
No ideas?

carldebilly, 2019-08-14:
Try to bring it closer to a router. I have one and the pairing worked on the first try.

way2many, 2019-08-17:
I have one of those and it was a pain to make it work correctly. Spent couple of hours trying to pair it correctly. After that you just hope that it will work correctly as, in my case, it only reports if there is smoke detected. No battery level and no link quality.

I test it from time to time with a burn out match.

closed 2019-10-23 by stale[bot]