Koenkk/zigbee2mqtt #1829: Update on hassio to 1.5.1 not working (Timeout | ERR! code ELIFECYCLE)

MrMennens, 2019-08-07:
# Bug Report
https://pastebin.com/zKk3Guqi
## What happened
Updated my zigbee2mqtt to the newest version (1.5.1)

## What did you expect to happen
A running version of zigbee2mqtt

## How to reproduce it (minimal and precise)
Just pressing start on the hassio add-on
Tried the dev add-on, works for 20 seconds, can turn my lamp on-off for 2/3 times then shuts down.

## Debug Info
zigbee2mqtt version:
zigbee2mqtt version 1.5.1 (commit #unknown)
CC253X firmware version:
CC2531_DEFAULT_20190608

Koenkk, 2019-08-08:
Did you read the upgrading notes? https://github.com/danielwelch/hassio-zigbee2mqtt#warning-breaking-changes-in-version-151.

As this is an hassio issue, please create it here: https://github.com/danielwelch/hassio-zigbee2mqtt/issues

closed 2019-08-08 by Koenkk

MrMennens, 2019-08-09:
Thanks!
Fixed it with new firmware of the cc2351.
Network key also updated.

Now trying to figure out how to start over.