Koenkk/zigbee2mqtt #1831: New zigbeeModel on supported device

acc, 2019-08-08:
I just installed a Nue/3A HGZB-01A controller and when I connected it to zigbee2mqtt it was listed as an unsupported model (LXN56-LC27LX1.1), even though I can see it listed as a supported device on the website.

I edited node_modules/zigbee-shepherd-converters/devices.js and changed:

`
{
        zigbeeModel: ['FNB56-ZSW23HG1.1'],
        model: 'HGZB-01A',
        vendor: 'Nue / 3A',
        description: 'Smart light controller',
        extend: generic.light_onoff_brightness,
    },
`

to 

`
{
        zigbeeModel: ['FNB56-ZSW23HG1.1', 'LXN56-LC27LX1.1'],
        model: 'HGZB-01A',
        vendor: 'Nue / 3A',
        description: 'Smart light controller',
        extend: generic.light_onoff_brightness,
    },
`

It works correctly although I would note that it registers in HA as a light even though it is really a switch (albeit an inline one).

acc, 2019-08-08:
One other note; this controller does not support brightness however it has a brightness slider in HA.

acc, 2019-08-08:
Actually found if I add my model number to : 

`
{
        zigbeeModel: ['FB56+ZSW05HG1.2', 'LXN56-LC27LX1.1'],
        model: 'HGZB-01A/02A',
        vendor: 'Nue / 3A',
        description: 'Smart 1 gang wall or in-wall switch',
        supports: 'on/off',
        fromZigbee: [fz.state, fz.ignore_onoff_change],
        toZigbee: [tz.on_off],
    },
`
I get a switch. It seems to also still have a light as part of the entity.

There seems to be a few instances of HGZB-01A and HGZB-02A in devices.js. There are stand alone lights and then there is this merged one. Not sure if this is a mistake or on purpose. My understand is that they are the same device except that the HGZB-02A supports brightness where as the HGZB-01A does not.

Koenkk, 2019-08-09:
Great, could you make a pull requests on https://github.com/koenkk/zigbee-shepherd-converters ?

acc, 2019-08-11:
Done

closed 2019-08-11 by acc