Koenkk/zigbee2mqtt #1832: Problem detected

Kryzek, 2019-08-08:
```
Starting zigbee2mqtt version 1.5.1 (commit #2107c9a)
and
Coordinator firmware version: '20190608'
```

This error appears every time device state changes:
```
PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
Error: Cluster command with key '0' and direction 'SERVER_TO_CLIENT' does not exist
    at Object.getSpecificCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/utils.js:103:15)
    at Function.parsePayloadCluster (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:174:31)
    at Function.parsePayload (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:167:25)
    at Function.fromBuffer (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:137:30)
    at Object.header (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:81:42)
    at dispatchIncomingMsg (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:528:29)
    at Controller.incomingMsgHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:625:12)
    at Controller.emit (events.js:198:13)
    at Object.bridge._areqEventBridge (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/event_bridge.js:11:16)
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/controller.js:104:20)
===== Buffer =====
{ type: 'Buffer', data: [ 25, 74, 0, 33, 0, 0, 23, 0, 0 ] }
===== zcl-packet header =====
{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":null,"seqNum":74,"cmdId":0}
===== zcl header =====
undefined
===============
```

I got this on my dev box when I tried to get Trust Contact Sensor to report it's battery levels
This is what I had in converters/devices.js:
```
    {
        zigbeeModel: ['CSW_ADUROLIGHT'],
        model: 'ZCTS-808',
        vendor: 'Trust',
        description: 'Wireless contact sensor',
        supports: 'contact',
          fromZigbee: [fz.ias_contact_dev_change, fz.ias_contact_status_change, fz.generic_battery_remaining],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
                (cb) => device.bind('genPowerCfg', coordinator, cb),
                (cb) => device.report('genPowerCfg','batteryPercentageRemaining', repInterval.MINUTE, repInterval.MAX, 1, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

Device seems to report battery level even when I get this error message posted above.
```
  zigbee2mqtt:debug 8/8/2019, 7:25:08 PM Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":200}}' of device 'CSW_ADUROLIGHT' (0x00158d0001cc7354) of endpoint 1
```

Otherwise device seems to be reporting states nicely. I do not have any other devices paired than this contact sensor.

Kryzek, 2019-08-09:
I updated to commit #d3a8a5c

Now with DEBUG=*

```
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg +1s
  unpi:parser --- parseNext [] +7ms
  serialport/stream _read reading +9ms
  serialport/binding-abstract read +9ms
  serialport/bindings/unixRead Starting read +9ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":58023,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":134,"securityuse":0,"timestamp":3911778,"transseqnumber":0,"len":9,"data":{"0":25,"1":60,"2":0,"3":33,"4":0,"5":0,"6":23,"7":0,"8":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":60,"cmdId":"statusChangeNotification","payload":{"zonestatus":33,"extendedstatus":0}}} +7ms
  zigbee2mqtt:debug 8/9/2019, 7:59:25 AM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'CSW_ADUROLIGHT' (0x00158d0001cc7354) of endpoint 1
  zigbee2mqtt:info 8/9/2019, 7:59:25 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cc7354', payload '{"contact":false,"linkquality":134,"battery":100}'
  zigbee-shepherd:af Message has unsupported cmdID: statusChangeNotification +4ms
  serialport/bindings/unixRead Finished read 34 bytes +4ms
  serialport/stream binding.read finished +4ms
  unpi:parser <-- [254,29,68,129,0,0,0,5,167,226,1,1,0,134,0,129,176,59,0,0,9,25,60,0,33,0,0,23,0,0,167,226,29,86] +6ms
  unpi:parser --- parseNext [254,29,68,129,0,0,0,5,167,226,1,1,0,134,0,129,176,59,0,0,9,25,60,0,33,0,0,23,0,0,167,226,29,86] +0ms
  unpi:parser --> parsed 29 - 2 - 4 - 129 - [�����;     <!��] - 86 +0ms
  znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":58023,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":134,"securityuse":0,"timestamp":3911809,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[25,60,0,33,0,0,23,0,0]}} +13ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":58023,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":134,"securityuse":0,"timestamp":3911809,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[25,60,0,33,0,0,23,0,0]}} +2ms

PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
Error: Cluster command with key '0' and direction 'SERVER_TO_CLIENT' does not exist
    at Object.getSpecificCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/utils.js:103:15)
    at Function.parsePayloadCluster (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:174:31)
    at Function.parsePayload (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:167:25)
    at Function.fromBuffer (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:137:30)
    at Object.header (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:81:42)
    at dispatchIncomingMsg (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:528:29)
    at Controller.incomingMsgHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:625:12)
    at Controller.emit (events.js:198:13)
    at Object.bridge._areqEventBridge (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/event_bridge.js:11:16)
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/controller.js:104:20)
===== Buffer =====
{ type: 'Buffer', data: [ 25, 60, 0, 33, 0, 0, 23, 0, 0 ] }
===== zcl-packet header =====
{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":null,"seqNum":60,"cmdId":0}
===== zcl header =====
undefined
===============

  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg +10ms
  unpi:parser --- parseNext [] +4ms
  serialport/stream _read reading +5ms
  serialport/binding-abstract read +9ms
  serialport/bindings/unixRead Starting read +5ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":58023,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":134,"securityuse":0,"timestamp":3911809,"transseqnumber":0,"len":9,"data":{"0":25,"1":60,"2":0,"3":33,"4":0,"5":0,"6":23,"7":0,"8":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":60,"cmdId":"statusChangeNotification","payload":{"zonestatus":33,"extendedstatus":0}}} +4ms
  zigbee2mqtt:debug 8/9/2019, 7:59:25 AM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'CSW_ADUROLIGHT' (0x00158d0001cc7354) of endpoint 1
  zigbee2mqtt:info 8/9/2019, 7:59:25 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cc7354', payload '{"contact":false,"linkquality":134,"battery":100}'
  zigbee-shepherd:af Message has unsupported cmdID: statusChangeNotification +2ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +3ms
  serialport/bindings/poller Polling for "readable" +22ms
  serialport/bindings/poller received "readable" +9s
  serialport/binding-abstract read +9s
  serialport/bindings/unixRead Starting read +9s
  serialport/bindings/unixRead Finished read 32 bytes +0ms
  serialport/stream binding.read finished +9s
  unpi:parser <-- [254,27,68,129,0,0,1,0,167,226,1,1,0,70,0,139,34,60,0,0,7,24,61,10,33,0,32,200,167,226,29,240] +9s
[... cut ...]
  zigbee2mqtt:info 8/9/2019, 8:00:14 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cc7354', payload '{"contact":true,"linkquality":131,"battery":100}'
  zigbee-shepherd:af Message has unsupported cmdID: statusChangeNotification +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +2ms
  serialport/bindings/poller Polling for "readable" +6ms
```  

There is also messages ```zigbee-shepherd:af Message has unsupported cmdID: statusChangeNotification``` with debugging on.

Koenkk, 2019-08-12:
I've added some more logging, can you update to the latest dev branch and provide the logging again?

Kryzek, 2019-08-13:
Will try after I get back to work.

Kryzek, 2019-08-13:
Here is the log with DEBUG=* commit #a62d6aa

```
PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
AssertionError [ERR_ASSERTION]: { frameCntl:
   { frameType: 1, manufSpec: 0, direction: 1, disDefaultRsp: 1 },
  manufCode: null,
  seqNum: 91,
  cmdId: 0 } deepEqual undefined
    at Object.header (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:88:20)
    at dispatchIncomingMsg (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:528:29)
    at Controller.incomingMsgHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:625:12)
    at Controller.emit (events.js:198:13)
    at Object.bridge._areqEventBridge (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/event_bridge.js:11:16)
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/controller.js:104:20)
    at Znp.emit (events.js:198:13)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/znp/znp.js:72:18)
    at Parser.emit (events.js:198:13)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/unpi/parser.js:39:26)
===== Buffer =====
{ type: 'Buffer', data: [ 25, 91, 0, 33, 0, 0, 23, 0, 0 ] }
===== zcl-packet header =====
{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":null,"seqNum":91,"cmdId":0}
===== zcl header =====
undefined
===============

  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg +204ms
  zigbee-herdsman:unpi:parser --- parseNext [] +11ms
  serialport/stream _read reading +12ms
  serialport/binding-abstract read +13ms
  serialport/bindings/unixRead Starting read +12ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":60505,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":131,"securityuse":0,"timestamp":2330794,"transseqnumber":0,"len":9,"data":{"0":25,"1":91,"2":0,"3":33,"4":0,"5":0,"6":23,"7":0,"8":0},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":91,"cmdId":"statusChangeNotification","payload":{"zonestatus":33,"extendedstatus":0}}} +11ms
  zigbee2mqtt:debug 8/13/2019, 3:50:22 PM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'CSW_ADUROLIGHT' (0x00158d0001cc7354) of endpoint 1
  zigbee2mqtt:info 8/13/2019, 3:50:22 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cc7354', payload '{"contact":false,"linkquality":131,"battery":100}'
  zigbee-shepherd:af Message has unsupported cmdID: statusChangeNotification +2ms
  serialport/bindings/unixRead Finished read 34 bytes +2ms
  serialport/stream binding.read finished +2ms
  zigbee-herdsman:unpi:parser <-- [254,29,68,129,0,0,0,5,89,236,1,1,0,131,0,199,144,35,0,0,9,25,91,0,33,0,0,23,0,0,89,236,29,74] +3ms
  zigbee-herdsman:unpi:parser --- parseNext [254,29,68,129,0,0,0,5,89,236,1,1,0,131,0,199,144,35,0,0,9,25,91,0,33,0,0,23,0,0,89,236,29,74] +0ms
  zigbee-herdsman:unpi:parser --> parsed 29 - 2 - 4 - 129 - [Y��ǐ#	[!Y�] - 74 +0ms
  zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1280,"srcaddr":60505,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":131,"securityuse":0,"timestamp":2330823,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[25,91,0,33,0,0,23,0,0]}} +14ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":1280,"srcaddr":60505,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":131,"securityuse":0,"timestamp":2330823,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[25,91,0,33,0,0,23,0,0]}} +1ms

PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
AssertionError [ERR_ASSERTION]: { frameCntl:
   { frameType: 1, manufSpec: 0, direction: 1, disDefaultRsp: 1 },
  manufCode: null,
  seqNum: 91,
  cmdId: 0 } deepEqual undefined
    at Object.header (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:88:20)
    at dispatchIncomingMsg (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:528:29)
    at Controller.incomingMsgHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:625:12)
    at Controller.emit (events.js:198:13)
    at Object.bridge._areqEventBridge (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/event_bridge.js:11:16)
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/controller.js:104:20)
    at Znp.emit (events.js:198:13)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/znp/znp.js:72:18)
    at Parser.emit (events.js:198:13)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/unpi/parser.js:39:26)
===== Buffer =====
{ type: 'Buffer', data: [ 25, 91, 0, 33, 0, 0, 23, 0, 0 ] }
===== zcl-packet header =====
{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":null,"seqNum":91,"cmdId":0}
===== zcl header =====
undefined
===============

  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg +8ms
  zigbee-herdsman:unpi:parser --- parseNext [] +4ms
  serialport/stream _read reading +5ms
  serialport/binding-abstract read +7ms
  serialport/bindings/unixRead Starting read +5ms
[... cut ...]
  zigbee2mqtt:debug 8/13/2019, 3:50:22 PM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'CSW_ADUROLIGHT' (0x00158d0001cc7354) of endpoint 1
  zigbee2mqtt:info 8/13/2019, 3:50:22 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cc7354', payload '{"contact":false,"linkquality":131,"battery":100}'
  zigbee-shepherd:af Message has unsupported cmdID: statusChangeNotification +2ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +3ms
  serialport/bindings/poller Polling for "readable" +23ms

```

Koenkk, 2019-08-13:
Should be fixed in latest dev, can you verify?

Kryzek, 2019-08-14:
Yes, latest dev (#22bed9a) fixed the issue. Thank you @Koenkk 

I'll make PR to get battery readings for the device now :)
```
  zigbee2mqtt:info 8/14/2019, 7:48:42 AM Successfully configured 0x00158d0001cc7354 (0x00158d0001cc7354)
  zigbee2mqtt:debug 8/14/2019, 7:48:50 AM Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":200}}' of device 'CSW_ADUROLIGHT' (0x00158d0001cc7354) of endpoint 1
  zigbee2mqtt:info 8/14/2019, 7:48:50 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cc7354', payload '{"contact":true,"linkquality":94,"battery":200}'
  zigbee2mqtt:debug 8/14/2019, 7:48:56 AM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'CSW_ADUROLIGHT' (0x00158d0001cc7354) of endpoint 1
  zigbee2mqtt:info 8/14/2019, 7:48:56 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cc7354', payload '{"contact":false,"linkquality":55,"battery":200}'
  zigbee2mqtt:debug 8/14/2019, 7:48:56 AM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":33}' of device 'CSW_ADUROLIGHT' (0x00158d0001cc7354) of endpoint 1
  zigbee2mqtt:info 8/14/2019, 7:48:56 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cc7354', payload '{"contact":false,"linkquality":57,"battery":200}'
```

Koenkk, 2019-08-14:
Great! thanks!

closed 2019-08-14 by Koenkk