Koenkk/zigbee2mqtt #1833: Suddenly a second coordinator appeared (0xffffffffffffffff)

johnnymijnhout, 2019-08-08:
# Bug Report
## What happened
Since yesterday, a second coordinator appeared in my Zigbee network. This is impossible as i only have 1 coordinator stick. It has a really weird id (0xffffffffffffffff)
## What did you expect to happen
My network should only have 1 coordinator
## How to reproduce it (minimal and precise)
I have no idea. It was running great for some weeks, and suddenly this second coordinator was there.

I also cannot restart zigbee2mqtt before i:
- Delete the 0xffffffffffffffff device from devices.yaml
- Delete all entries of 0xffffffffffffffff in database.db

If i generate the network map, it immediately appears again after i removed it.

I tried to "ban" the device too in the config, but in my map, it still reappears. Also, all devices connected to the real coordinator are also connected to the "fake coordinator", which makes me think it is a clone of the real coordinator.
## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: 20190608 DEFAULT

## Some log snippets:
`8/8/2019, 8:47:10 PM - debug: Accept device incoming with ieeeAddr '0xffffffffffffffff' permit join is 'false'
8/8/2019, 8:47:10 PM - info: Banned device tried to connect (0xffffffffffffffff)
8/8/2019, 8:47:25 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0xffffffffffffffff"' of device 'undefined' (0xffffffffffffffff) of endpoint 1`

`8/8/2019, 8:50:58 PM - info: New device 'undefined' with address 0xffffffffffffffff connected!
8/8/2019, 8:50:58 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xffffffffffffffff","meta":{}}'
8/8/2019, 8:51:00 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0xffffffffffffffff"' of device 'undefined' (0xffffffffffffffff) of endpoint 1`

`8/8/2019, 8:53:11 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0xffffffffffffffff"' of device 'undefined' (0xffffffffffffffff) of endpoint 1
8/8/2019, 8:53:11 PM - warn: Device with modelID 'undefined' is not supported.
8/8/2019, 8:53:11 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

[image]

Koenkk, 2019-08-09:
Is this device also logged on startup?

johnnymijnhout, 2019-08-09:
It does not appear in the list of:
`8/9/2019, 4:51:42 PM - info: Currently 21 devices are joined:`

But pretty much immediately after that:
```
 8/9/2019, 4:52:19 PM - info: New device 'undefined' with address 0xffffffffffffffff connected!
8/9/2019, 4:52:19 PM - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
8/9/2019, 4:52:19 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0xffffffffffffffff","meta":{}}'
8/9/2019, 4:52:19 PM - warn: Device with modelID 'undefined' is not supported.
8/9/2019, 4:52:19 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```
Then i see this around every 15s:
```8/9/2019, 4:56:41 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0xffffffffffffffff"' of device 'undefined' (0xffffffffffffffff) of endpoint 1```

johnnymijnhout, 2019-08-09:
Also, my closest neighbour lives 100m from my house, so it cannot be interference with other coordinators from neighbours.

johnnymijnhout, 2019-08-10:
Today i erased and reflashed the CC2531 with the source routing firmware, only to discover that immediately the ffffffffff device pops up again :(

Koenkk, 2019-08-12:
`8/9/2019, 4:56:41 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0xffffffffffffffff"' of device 'undefined' (0xffffffffffffffff) of endpoint 1` makes me think that this is actually a device on the network, do you see any pointers when sniffing? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

johnnymijnhout, 2019-08-13:
I will try this, but if this is true, it would be very weird that it has the exact same device connections as the original coordinator has :S

johnnymijnhout, 2019-08-13:
I was checking the log, and i noticed i didn't see the fffffff notifications anymore. After removing the device again from database.db and devices.yaml, and restarting zigbee2mqtt, it was really gone. 
[image]
I have no idea why this happened, i still have the same devices in my network. Wondering for how long this will last...

johnnymijnhout, 2019-08-15:
Everything seems to be back to normal, will close this issue now as i seem to be the only one, and i cannot reproduce it anymore

closed 2019-08-15 by johnnymijnhout