Koenkk/zigbee2mqtt #1835: Dummy serial port

xoxys, 2019-08-08:
I use ansible to deploy and manage my zigbee2mqtt + mosquitto + hassio setup and for each ansible role I have implemented ci driven deployment tests. These deployment tests are running on temporary Amazon EC2 machines which (naturally) don't have a working zigbee device. That's the reason why deployment tests (which also includes stating systemd daemon) will fail all the time.

Would it be possible to implement something like that:
```
serial:
  port: "dummy"
```

So zigbee2mqtt will start without errors? I know this may be a very small corner case and if you don't want to have it in feel free to close the issue :)

xoxys, 2019-10-09:
Keep. @Koenkk if possible :)

closed 2020-02-14 by stale[bot]

wroadd, 2025-06-19:
I'm using the same config (with HA) in AWS EC2 for education purposes, and a dummy adapter could help to show the UI and just see the installation method.