Koenkk/zigbee2mqtt #1838: Zigbee2mqtt broken after HA and Hassbian upgrade

ThePirat91, 2019-08-09:
# Bug Report
## What happened
Hello, yesterday I upgraded my HA from 0.94.5 to 0.97 and my hassbian (on an rpi 3 b+) to latest versions.
I have also had to update raspbian to latest version (from stretch to buster).
After this upgrades all works but zigbee2mqtt.
I had version 1.0.1 of the addon, installed through hassbian-scripts, and everything worked well.
After raspian upgrade I have been starting to see some errors about node.js and the addon refuses to start.
Please see below for node.js log (zigbee2mqtt does log nothing cause it cannot actually start)

-- Logs begin at Fri 2019-08-09 12:53:42 CEST. --
Aug 09 16:21:00 hassbian npm[1645]: npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
Aug 09 16:21:00 hassbian npm[1645]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Aug 09 16:21:12 hassbian npm[1645]: npm ERR! A complete log of this run can be found in:
Aug 09 16:21:12 hassbian npm[1645]: npm ERR!     /home/pi/.npm/_logs/2019-08-09T14_20_58_721Z-debug.log
Aug 09 16:21:12 hassbian systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Aug 09 16:21:12 hassbian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Aug 09 16:21:12 hassbian systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Aug 09 16:21:12 hassbian systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 74.
Aug 09 16:21:12 hassbian systemd[1]: Stopped zigbee2mqtt.
Aug 09 16:21:12 hassbian systemd[1]: Started zigbee2mqtt.
Aug 09 16:22:45 hassbian npm[1695]: > zigbee2mqtt@1.0.1 start /opt/zigbee2mqtt
Aug 09 16:22:45 hassbian npm[1695]: > node index.js
Aug 09 16:22:45 hassbian npm[1695]:                 Zigbee2mqtt requires node version 8.11 - 10, you are running v11.15.0!
Aug 09 16:22:47 hassbian npm[1695]:   zigbee2mqtt:info 8/9/2019, 4:22:47 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-08-09.16-22-46'
Aug 09 16:22:48 hassbian npm[1695]: /opt/zigbee2mqtt/node_modules/bindings/bindings.js:88
Aug 09 16:22:48 hassbian npm[1695]:         throw e
Aug 09 16:22:48 hassbian npm[1695]:         ^
Aug 09 16:22:48 hassbian npm[1695]: Error: The module '/opt/zigbee2mqtt/node_modules/serialport/build/Release/serialport.node'
Aug 09 16:22:48 hassbian npm[1695]: was compiled against a different Node.js version using
Aug 09 16:22:48 hassbian npm[1695]: NODE_MODULE_VERSION 57. This version of Node.js requires
Aug 09 16:22:48 hassbian npm[1695]: NODE_MODULE_VERSION 67. Please try re-compiling or re-installing
Aug 09 16:22:48 hassbian npm[1695]: the module (for instance, using `npm rebuild` or `npm install`).
Aug 09 16:22:48 hassbian npm[1695]:     at Object.Module._extensions..node (internal/modules/cjs/loader.js:857:18)
Aug 09 16:22:48 hassbian npm[1695]:     at Module.load (internal/modules/cjs/loader.js:685:32)
Aug 09 16:22:48 hassbian npm[1695]:     at Function.Module._load (internal/modules/cjs/loader.js:620:12)
Aug 09 16:22:48 hassbian npm[1695]:     at Module.require (internal/modules/cjs/loader.js:723:19)
Aug 09 16:22:48 hassbian npm[1695]:     at require (internal/modules/cjs/helpers.js:14:16)
Aug 09 16:22:48 hassbian npm[1695]:     at bindings (/opt/zigbee2mqtt/node_modules/bindings/bindings.js:81:44)
Aug 09 16:22:48 hassbian npm[1695]:     at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/serialport/lib/bindings/linux.js:2:36)
Aug 09 16:22:48 hassbian npm[1695]:     at Module._compile (internal/modules/cjs/loader.js:816:30)
Aug 09 16:22:48 hassbian npm[1695]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
Aug 09 16:22:48 hassbian npm[1695]:     at Module.load (internal/modules/cjs/loader.js:685:32)
Aug 09 16:22:48 hassbian npm[1695]: npm ERR! code ELIFECYCLE
Aug 09 16:22:48 hassbian npm[1695]: npm ERR! errno 1
Aug 09 16:22:48 hassbian npm[1695]: npm ERR! zigbee2mqtt@1.0.1 start: `node index.js`
Aug 09 16:22:48 hassbian npm[1695]: npm ERR! Exit status 1
Aug 09 16:22:48 hassbian npm[1695]: npm ERR!

The problem is the node.js version installed compared to the version the addon supports.

Do you something similar?
How can we solve?

Thanks
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.01
CC253X firmware version: 20180815

timstanley1985, 2019-08-09:
You need to use node v10. You are using node 11

timstanley1985, 2019-08-09:
From your log: Aug 09 16:22:45 hassbian npm[1695]: Zigbee2mqtt requires node version 8.11 - 10, you are running v11.15.0!

ThePirat91, 2019-08-10:
Thanks for the answer.
I know the problem is the node version (wrote it also on my first post, end part) but the problem is....how can I handle this?
Can I uninstall the last version and install a previous one on hassbian? If so, how?
Thanks

ThePirat91, 2019-10-11:
I'm gonna close the ticket. I solved my issue installing node v.10 instead of v.11
Thank you.

closed 2019-10-11 by ThePirat91