Koenkk/zigbee2mqtt #1840: Zigbee 2 Bluetooth ?

podarok, 2019-08-09:
I was trying to run zigbee2mqtt in Android phone
- termux + termux api
- ubuntu in chroot
And compilation went well.
The only complexity is usb is not directly accessible via /dev/ttyXxxx
But Bluetooth - accessible.

Is there a chance to extend cc2531 firmware to send zigbee traffic to Bluetooth?
In this case we would be able to run dongle from power bank and connect to it via Bluetooth from central hub where zigbee2mqtt installed.

Koenkk, 2019-08-10:
The CC2531 doesnt have bluetooth maybe it would be possible for the CC1352P but for now this is out of scope.

JLFN, 2019-08-13:
@podarock esp8266 with cc2530 and Vpn hotspot will make it.

closed 2019-10-19 by stale[bot]