Koenkk/zigbee2mqtt #1841: Newer docker container builds crash every 3 or 4 minutes

BudBundi, 2019-08-10:
#1267 Bug Report
## What happened
zigbee2mqtt crash in reporting setup phase and restart, often after restart all devices are offline

## What did you expect to happen
a stable run

## How to reproduce it (minimal and precise)
just start with new container
last known good, that has no problems for me 
sha256:ca344e34e0f04d8338690b6993747a863171da7e1640f228aad601a04fe648cd

Created | 2019-08-05 20:29:33

## Debug Info
zigbee2mqtt version: sha256:942ca9df5764229bbfff2bdb0311ab9dc43621f039f44b1bedcd2ed65e522387 
Created | 2019-08-08 20:37:54
CC253X firmware version:
CC2531_SOURCE_ROUTING_20190706

crash log:
https://pastebin.com/GJm4SDbw

BudBundi, 2019-08-11:
todays koenkk/zigbee2mqtt:arm32v6-dev image haven't fix it, still the message

> /app/node_modules/zigbee-shepherd-converters/converters/toZigbee.js:1447
            if (message.hasOwnProperty('transition')) {
                        ^
TypeError: Cannot read property 'hasOwnProperty' of null
    at Object.convert (/app/node_modules/zigbee-shepherd-converters/converters/toZigbee.js:1447:25)

my feeling is that it most often crash after 

> 8/11/2019, 11:54:59 AM - info: MQTT publish: topic 'zigbee2mqtt/Badezimmer', payload '{"state":"OFF","brightness":254,"color_temp":290,"color":{"x":0.301,"y":0.116}}'

which is a GLEDOPTO controller

Koenkk, 2019-08-12:
Can you provide the debug logging of the crash?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

BudBundi, 2019-08-12:
Sure.
The logs from the docker container from the mounted directory:
https://pastebin.com/V89uwYGn
https://pastebin.com/YuCDF0v1
https://pastebin.com/v8jCedJH

NPM logs from mounted directory:
https://pastebin.com/J0XwgQ11
https://pastebin.com/UqUqCwzr

I see nothing new in these logs, if I look at the container over portainer, still the same message:
https://pastebin.com/GghH303R

Koenkk, 2019-08-14:
I would expect that the logs you provided also contain the crash. In this way I cannot see what message caused the crash.

BudBundi, 2019-08-14:
I tried a few things.
Switch to Z_Stack 3 on a CC2652R, after repairing all lights, still crash with newer image.
Back on a CC2531an Z Stack 1.2, still crash with new image, old one still good.

I extract the log over portainer with debug on, maybe it contains better information for you
https://pastebin.com/FWaft8WD

Is there another directory with logs that I should mount? Or another thing I should try?

BudBundi, 2019-08-14:
I'm just guessing but maybe this https://github.com/Koenkk/zigbee-shepherd-converters/commit/54a9928d692d0963b2a52fb33f454191ce1bd896 change cause this?

BudBundi, 2019-08-15:
I just looked witch versions of zigbee-shepherd the good image use it is 10.2.0, so a release after the mentioned change. The good image use zigbee-herdsman@0.4.5, I try to find a change in herdsman witch may case this problem

BudBundi, 2019-08-15:
Sorry my bad, again. The package.json from the running image say "zigbee-herdsman": "0.5.2", "zigbee-shepherd-converters": "10.2.0"
so it seems to build after commit #a604fd63451fbda677a7da97d62af34b5e9ea99f

Koenkk, 2019-08-16:
I found the issue, will publish a fix soon.

Koenkk, 2019-08-16:
Fixed in latest dev, thanks!

closed 2019-08-16 by Koenkk

BudBundi, 2019-08-17:
The travis build for this failed, so there is no new image. Is it possible to rerun it?

Koenkk, 2019-08-18:
Fixed now