Koenkk/zigbee2mqtt #1842: Setting sensitivity on DJT11LM - MAC transaction expired

tiagofreire-pt, 2019-08-10:
Using this method: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#xiaomi-aqara-vibration-sensor-djt11lm

I get errors on setting the sensitivity:

zigbee2mqtt:info 8/10/2019, 11:42:14 AM Zigbee publish to device '0x00158d000312c3d6', genBasic - write - [{"attrId":65293,"dataType":32,"attrData":1}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null zigbee2mqtt:error 8/10/2019, 11:42:21 AM Zigbee publish to device '0x00158d000312c3d6', genBasic - write - [{"attrId":65293,"dataType":32,"attrData":1}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null failed with error Error: AF data request fails, status code: 240. MAC transaction expired. zigbee2mqtt:info 8/10/2019, 11:42:21 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 240. MAC transaction expired.","meta":{"entity":{"ID":"0x00158d000312c3d6","type":"device","friendlyName":"0x00158d000312c3d6"},"message":"{\n \"sensitivity\": \"high\" \n}"}}' zigbee2mqtt:info 8/10/2019, 11:42:21 AM Zigbee publish to device '0x00158d000312c3d6', genBasic - write - [{"attrId":65293,"dataType":32,"attrData":1}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null zigbee2mqtt:info 8/10/2019, 11:42:22 AM Zigbee publish to device '0x00158d000312c3d6', genBasic - write - [{"attrId":65293,"dataType":32,"attrData":1}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null

dzungpv, 2019-08-10:
Do you use any router?

tiagofreire-pt, 2019-08-10:
Yes, I do. But this device is bind directly to the coordinator, as shown on zigbee2mqtt map component in Home Assistant.

dzungpv, 2019-08-10:
Zigbee map not all true, i advice you remove router from the network and retry, if it work the problem may cause by router

tiagofreire-pt, 2019-08-13:
I understand, but without the router I can't get communication as the signal is quite weak.

closed 2019-10-19 by stale[bot]