Koenkk/zigbee2mqtt #1845: No Network Route multiple times a day

florian-ka, 2019-08-11:
Hey Guys,

i'm very happy with zigbee2mqtt when the Network Routes are available. But since the beginning i have problem with the connectivity to the devices.

I have 18 devices (Hue lamps, Aqara Motion Sensor, Ikea lamps). But multiple time  a day the whole zigbee network seems to crash / corrupt. Sometimes only a few but mostly the whole day / night. Sometimes it fixes itself if i just wait and sometimes stopping Z2M and replugging the CC2531 and then restarting Z2M is fixing it.

Hue Lightstrip error:
`Zigbee publish to device 'Hue Lightstrip 1', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.`

This error happens for multiple devices. I tried a extension cord but even when the device is 1 meter away i have this error message every so often.

zigbee2mqtt version: latest 1.5.1
CC2531 firmware version: 20190608

If you need more information about my setup or my configuration just ask and i will happily provide it.

Greetz

Koenkk, 2019-08-12:
The source routing firmware might fix the problem: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

florian-ka, 2019-08-12:
Thanks will try that but since my post my zigbee2mqtt does not get any data from any device anymore. is there a way to "reset" the network without deleting and repairing the devices ?

Koenkk, 2019-08-13:
@florian-ka just reflash the stick with that firmware, you don't have to re-pair devices after that.

florian-ka, 2019-08-13:
still no device had connection. after i delete the database and repaired the devices they work. Looking at the stability with the source routing firmware now for the next few days.

Will report back

hwijers, 2019-08-14:
I also had problems with:
zigbee2mqtt version: latest 1.5.1
CC2531 firmware version: 20190608
I finally went back to firmware version:
20190425 with source routing disabled
This firmware version works best for me.

closed 2019-10-20 by stale[bot]