Koenkk/zigbee2mqtt #1851: AssertionError .. deepEqual undefined

kiall, 2019-08-12:
# Bug Report
## What happened

Stacktrace when pressing Ikea E1524 button

## What did you expect to happen

No stacktrace :)

## How to reproduce it (minimal and precise)

Update to dev, press button on Ikea E1524 remote!

I'm running the docker "latest-dev" image - I believe this is based off a62d6aad91a948220b4e1b2031c14d8472e94190, however the logs just show 1.5.1 (commit #unknown).

## Debug Info

zigbee2mqtt version: 
CC253X firmware version:

```zigbee2mqtt    | PROBLEM DETECTED
zigbee2mqtt    | Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
zigbee2mqtt    | ===== Stack =====
zigbee2mqtt    | AssertionError [ERR_ASSERTION]: { frameCntl:
zigbee2mqtt    |    { frameType: 1, manufSpec: 0, direction: 0, disDefaultRsp: 0 },
zigbee2mqtt    |   manufCode: null,
zigbee2mqtt    |   seqNum: 79,
zigbee2mqtt    |   cmdId: 2 } deepEqual undefined
zigbee2mqtt    |     at Object.header (/app/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:88:20)
zigbee2mqtt    |     at dispatchIncomingMsg (/app/node_modules/zigbee-herdsman/dist/lib/components/af.js:528:29)
zigbee2mqtt    |     at Controller.incomingMsgHandler (/app/node_modules/zigbee-herdsman/dist/lib/components/af.js:625:12)
zigbee2mqtt    |     at Controller.emit (events.js:182:13)
zigbee2mqtt    |     at Object.bridge._areqEventBridge (/app/node_modules/zigbee-herdsman/dist/lib/components/event_bridge.js:11:16)
zigbee2mqtt    |     at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/lib/components/controller.js:104:20)
zigbee2mqtt    |     at Znp.emit (events.js:182:13)
zigbee2mqtt    |     at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/dist/znp/znp.js:72:18)
zigbee2mqtt    |     at Parser.emit (events.js:182:13)
zigbee2mqtt    |     at Parser.parseNext (/app/node_modules/zigbee-herdsman/dist/unpi/parser.js:39:26)
zigbee2mqtt    | ===== Buffer =====
zigbee2mqtt    | { type: 'Buffer', data: [ 1, 79, 2 ] }
zigbee2mqtt    | ===== zcl-packet header =====
zigbee2mqtt    | {"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":0},"manufCode":null,"seqNum":79,"cmdId":2}
zigbee2mqtt    | ===== zcl header =====
zigbee2mqtt    | undefined
zigbee2mqtt    | ===============
```

Koenkk, 2019-08-13:
Should be fixed in latest dev, can you verify?

kiall, 2019-08-14:
Thanks! Will check this evening once home..

kiall, 2019-08-14:
Looks good now!

closed 2019-08-14 by kiall

Ulli2k, 2019-08-15:
I just repulled the docker container but I still have the following issue
`
===== zcl =====,
{"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":45,"cmdId":"tradfriArrowSingle","payload":{"value":256,"value2":13}},
===============,
,
,
,
,
,
,
  zigbee2mqtt:info 8/15/2019, 2:01:53 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe177cc9', payload '{"battery":60,"linkquality":5,"action":"arrow_right_click"}',
  zigbee2mqtt:info 8/15/2019, 2:01:54 PM MQTT publish: topic 'zigbee2mqtt/0x00124b001d3b064a', payload '{"state":true,"linkquality":15,"led_state":false}',
  zigbee2mqtt:info 8/15/2019, 2:01:56 PM Successfully configured 0x000d6ffffe177cc9 (0x000d6ffffe177cc9),
,
,
,
,
,
,
PROBLEM DETECTED,
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues,
===== Stack =====,
AssertionError [ERR_ASSERTION]: { frameCntl:,
   { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },,
  manufCode: 4476,,
  seqNum: 47,,
  cmdId: 'tr... deepEqual { frameCntl:,
   { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },,
  manufCode: 4476,,
  seqNum: 47,,
  cmdId: 'tr...,
    at cb (/app/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:56:24),
    at /app/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/zcl.js:31:17,
    at Dissolve.<anonymous> (/app/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/functional.js:76:17),
    at Object.onceWrapper (events.js:273:13),
    at Dissolve.emit (events.js:182:13),
    at Dissolve.<anonymous> (/app/node_modules/zigbee-herdsman/dist/deprecated/dissolve-chunks/index.js:65:29),
    at Dissolve.emit (events.js:182:13),
    at emitReadable_ (/app/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:456:10),
    at emitReadable (/app/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:450:7),
    at readableAddChunk (/app/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:206:11),
===== Buffer =====,
{ type: 'Buffer', data: [ 5, 124, 17, 47, 7, 1, 1, 13, 0 ] },
===== Arguments =====,
<Buffer 05 7c 11 2f 07 01 01 0d 00> 5,
===== zcl-packet parse =====,
{"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":47,"cmdId":"tradfriArrowSingle","payload":{"value":257}},
===== zcl =====,
{"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":47,"cmdId":"tradfriArrowSingle","payload":{"value":257,"value2":13}},
===============,
,
,
,
,
,
,
  zigbee2mqtt:info 8/15/2019, 2:02:07 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe177cc9', payload '{"battery":60,"linkquality":0,"action":"arrow_left_click"}',
,
,
,
,
,
,
PROBLEM DETECTED,
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues,
===== Stack =====,
AssertionError [ERR_ASSERTION]: { frameCntl:,
   { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },,
  manufCode: 4476,,
  seqNum: 48,,
  cmdId: 'tr... deepEqual { frameCntl:,
   { frameType: 1, manufSpec: 1, direction: 0, disDefaultRsp: 0 },,
  manufCode: 4476,,
  seqNum: 48,,
  cmdId: 'tr...,
    at cb (/app/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:56:24),
    at /app/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/zcl.js:31:17,
    at Dissolve.<anonymous> (/app/node_modules/zigbee-herdsman/dist/deprecated/zcl-packet/lib/functional.js:76:17),
    at Object.onceWrapper (events.js:273:13),
    at Dissolve.emit (events.js:182:13),
    at Dissolve.<anonymous> (/app/node_modules/zigbee-herdsman/dist/deprecated/dissolve-chunks/index.js:65:29),
    at Dissolve.emit (events.js:182:13),
    at emitReadable_ (/app/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:456:10),
    at emitReadable (/app/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:450:7),
    at readableAddChunk (/app/node_modules/zigbee-herdsman/node_modules/readable-stream/lib/_stream_readable.js:206:11),
===== Buffer =====,
{ type: 'Buffer', data: [ 5, 124, 17, 48, 7, 0, 1, 13, 0 ] },
===== Arguments =====,
<Buffer 05 7c 11 30 07 00 01 0d 00> 5,
===== zcl-packet parse =====,
{"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":48,"cmdId":"tradfriArrowSingle","payload":{"value":256}},
===== zcl =====,
{"frameCntl":{"frameType":1,"manufSpec":1,"direction":0,"disDefaultRsp":0},"manufCode":4476,"seqNum":48,"cmdId":"tradfriArrowSingle","payload":{"value":256,"value2":13}},
===============,
,
,
,
,
,
,
  zigbee2mqtt:info 8/15/2019, 2:02:13 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe177cc9', payload '{"battery":60,"linkquality":39,"action":"arrow_right_click"}',
`

kiall, 2019-08-15:
@Ulli2k Did you recreate the container, or just restart it?

Ulli2k, 2019-08-15:
delete and recreate

Koenkk, 2019-08-16:
@Ulli2k this one looks ok, I will make some changes that it doesn't appear anymore.

Koenkk, 2019-08-16:
@Ulli2k should be fixed in latest dev.

Ulli2k, 2019-08-18:
looks good thanks!