Koenkk/zigbee2mqtt #1853: Trying to support new devide - Unrecognized Cluster

joe540, 2019-08-13:
# Bug Report
## What happened
I'm trying to add an unsoported device, the LivingWise LVS-ZBSC7 Scene Controller. I created a blank device configuration with the model ID took from the log.
When I press on a button of the switch, zigbee2mqtt crash with "Unrocognized Cluster" message
Device configuration:
  {
        zigbeeModel: ['75d430d66c164c26ac8601c05932dc94'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
        model: 'SCENECONTROLLER', // Vendor model number, look on the device for a model number
        vendor: 'LivingWise', // Vendor of the device (only used for documentation and startup logging)
        description: 'LivingWise Scene Controller', // Description of the device
supports: '7 scenes buttons', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},

Logs :
https://pastebin.com/bnTmJWEi
## What did you expect to happen
Unsupported command shown in the log, for me to integrate the device
## How to reproduce it (minimal and precise)
Pair the device, create a blank device configuration in device.js (as shown in https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html) and push on a button of the device
## Debug Info
zigbee2mqtt version: 1.5.1 (commit #ac3b924)
CC253X firmware version: 20190608

joe540, 2019-08-13:
OK, a little digging and I found that the device reports an unknown cluster (with id 23) and command 8

I was able to add it to the cluster definition

In /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl-id/definitions/cluster_defs.json
```
   "manuSpecificLivingWise": {
        "attrId": null,
        "cmd": null,
        "cmdRsp":  { "unkown": 8 }
    }
```
In /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl-id/definitions/common.json,
under "clusterId"
       ` "manuSpecificLivingWise" : 23,`

In /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl-packet/lib/defs/zcl_meta.json
```
      "manuSpecificLivingWise": {
            "unkown": {
                "params": [}.                ],
                "dir": 1
            }
        },
```
I'm probably missing parameters but I don't know them. The app doesn't crash anymore, but there's not reports of commands being received with no controller. What's the next step to discover what's missing, the params, etc?

device in the database:
`{"id":2,"type":"Router","ieeeAddr":"0x00124b0017fab5a4","nwkAddr":11064,"manufId":0,"manufName":"欧瑞博","powerSource":"Mains (single phase)","modelId":"75d430d66c164c26ac8601c05932dc94","epList":[1],"zclVersion":1,"appVersion":"v1.0.08_20160106","hwVersion":2,"status":"online","joinTime":1565687764,"endpoints":{"1":{"profId":260,"epId":1,"devId":12,"inClusterList":[0,5],"outClusterList":[0,5,23],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genScenes":{"dir":{"value":3},"attrs":{}},"manuSpecificLivingWise":{"dir":{"value":2},"attrs":{}}}}},"dateCode":"2016-01-06","_id":"p7kE7fKMAKzPkhaT"}`

joe540, 2019-08-13:
I also figured out that the "dev" build gives way more info (instead of crashing), here's the info:
```
 ===== Stack =====
AssertionError [ERR_ASSERTION]: { frameCntl:
   { frameType: 1, manufSpec: 0, direction: 1, disDefaultRsp: 1 },
  manufCode: null,
  seqNum: 30,
  cmdId: 8 } deepEqual undefined
    at Object.header (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:90:20)
    at dispatchIncomingMsg (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/lib/components/af.js:528:29)
    at Controller.incomingMsgHandler (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/lib/components/af.js:625:12)
    at Controller.emit (events.js:198:13)
    at Object.bridge._areqEventBridge (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/lib/components/event_bridge.js:11:16)
    at Znp.<anonymous> (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/lib/components/controller.js:104:20)
    at Znp.emit (events.js:198:13)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/znp/znp.js:72:18)
    at Parser.emit (events.js:198:13)
    at Parser.parseNext (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/unpi/parser.js:39:26)
===== Arguments =====
<Buffer 19 1e 08 03 00 00> 23
===== Buffer =====
{ type: 'Buffer', data: [ 25, 30, 8, 3, 0, 0 ] }
===== zcl-packet header =====
{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":null,"seqNum":30,"cmdId":8}
===== zcl header =====
undefined
===============
```

Note that in the "Arguments", only one number change depending on the button that is pressed:
`<Buffer 19 1e 08 XX 00 00> 23`
Where XX is the button number (01 to 07)
I basically just want to notify that the button has been pushed so I can handle it in my home automation software to do stuff!

Thanks

Koenkk, 2019-08-14:
Please use the dev branch to do this, with the dev branch you only have to define the cluster in one file: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts (instead of 3). 

Add a `console.log(msg.zclMsg.cmdId)` here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/lib/components/af.ts#L705 and add to https://github.com/Koenkk/zigbee-herdsman/blob/master/src/lib/components/af.ts#L706 what is logged.

joe540, 2019-08-14:
Hi!

   First, thanks for your time! I did create the cluster 23 with cmd Id 8 so that I don't have an error anymore. I did add the console.log but nothing is logged on button pressed (nothing from the added console.log, nor the regular "no converter" message I should see)

I know it receives a message, because if the cluster doesn't exists, it throws an error (with the data) each time I press a button

Here's my cluster:
```
manuSpecificLivingWise: {
        ID: 23,
        attributes: {},
        commands: {},
        commandsResponse: {
            sceneNotification: {
                ID: 8,
                parameters: [
                    { name: 'button', type: dataType_1.default.uint8 },
                    { name: 'button1', type: dataType_1.default.uint8 },
                    { name: 'button2', type: dataType_1.default.uint8 },
                ],
            },
        },
    },
```

I tried adding 6 uint8 parameters (because I have the following buffer: `{ type: 'Buffer', data: [ 25, 30, 8, 3, 0, 0 ] } ` but I get the following error if I have more than 24bit of parameters:

```
RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 5. Received 6
    at boundsError (internal/buffer.js:49:9)
    at Buffer.readUInt8 (internal/buffer.js:137:5)
    at BuffaloZcl.readUInt8 (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:34:35)
    at BuffaloZcl.read (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:264:25)
    at BuffaloZcl.read (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/zcl/buffaloZcl.js:263:26)
    at Function.parsePayloadCluster (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:173:47)
    at Function.parsePayload (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:153:25)
    at Function.fromBuffer (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:129:30)
    at Object.header (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:83:42)
    at dispatchIncomingMsg (/opt/zigbee2mqtt-dev/node_modules/zigbee-herdsman/dist/lib/components/af.js:528:29)
```

I dug a bit and the type of the message received is "incomingMsg" instead of "zclIncomingMsg"
so it never go in the "case" there: [https://github.com/Koenkk/zigbee-herdsman/blob/master/src/lib/components/af.ts#L665](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/lib/components/af.ts#L665)

Any idea?

Thanks Again
Jonathan

Koenkk, 2019-08-15:
Do you see anything when pressing the button when running with `DEBUG=*`.?

Side note: note that the current controller code is a mess and i'm currently refactoring it, it will probably land in the dev after zigbee2mqtt 1.6. With this it will probably be easier to add support (https://github.com/Koenkk/zigbee-herdsman/tree/controller).

joe540, 2019-08-16:
I didn't think about that one, but nop, there's nothing when I press the button

If I add a console.log in the "incomingMsg" case, for example on line 658 [https://github.com/Koenkk/zigbee-herdsman/blob/master/src/lib/components/af.ts#L658](https://github.com/Koenkk/zigbee-herdsman/blob/master/src/lib/components/af.ts#L658)
I do have a print on each button press
 
I do have a programming background but I don't know how to handle this massive job you did with this project, that's incredible. If there's nothing else to do in the meantime but to wait for the refactor
of the code, I'll wait. I found those zigbee scene controllers for a REALLY decent price, so it'll still be worth waiting :P

Koenkk, 2019-08-16:
If you don't see it know, the refactor won't solve it. It means that the CC2531 doesn't propagate the message. 

Next step would be to sniff the traffic to see what the device sends when pressing the button https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

joe540, 2019-08-21:
Isn't weird to have all the data in the error output when the cluster is not set?

I'll order a programmer and try to sniff it.

Thanks

closed 2019-10-27 by stale[bot]

codewise-nicolas, 2020-01-04:
I have the same livingwise controller, bought them on clearance.
Due to the problems with getting it to work on zigbee2mqtt and not knowing how to change the led, I ended up replacing the module onboard with ah esp8266 i coded myself, to work over mqtt directly (over wifi off course).

let me know if you would be interested in instructions/info/code to do it yourself and I will see about posting it somewhere.

joe540, 2020-01-04:
Hi @codewise-nicolas you for sure could post an how-to guide somewhere because this project seems really nice!

In my case, I tried again with Zigbee2mqtt a week ago because I saw there was lot of updates and that another Orvibo device with buttons was newly support. The Livingwise SC7 being just a rebranded version of Orvibo's one, I gave it a shot and made the device model and converter which basically just got merged by Koenkk today ([see it on the Zigbee-herdsman-converters repo](https://github.com/Koenkk/zigbee-herdsman-converters/pull/862).

All 7 buttons work (press, hold, release). I didn't try to reverse engineer the LEDs (as that would probably require an Orvibo/Livingwise hub and a sniffer to sniff the communication) but in my case, it's not that much of an issue as I use the 7 buttons as "actions" and not as "scenes"

Those scene controller being pretty cheap here in canada ([15$ CAN on prime cables](https://www.primecables.ca/p-368984-lvs-zb-sc7-neutral-zigbee-smart-7-button-scene-controller-us-type-neutral#sku381087)), while others I found (Z-Wave and or Zigbee) were 4-5 times more expensive...

Thanks for the time you took to propose your project. Your ESP8266 project could probably be usefull to multiple people (using relay boards for instance) so you should share it somewhere :)