Koenkk/zigbee2mqtt #1856: TRV new model

zhapwey, 2019-08-13:
# Bug Report
## What happened
I have seen that we have several models of thermostatic valve for radiator, but ecozy does not seem to answer and eurotric is expensive, I have ordered one from alibaba (https://www.gmail.alibaba.com/product-detail/digital-programmable-trv-thermostatic -radiator-valve-and-zigbee-water-radiator-thermostat-60760050567.html? spm = a2700.8699010.normalList. 22.3172648egkRITS)

I have little idea how to add the device but really want to do it
## What did you expect to happen
I created the device in devices.js but I receive an error in
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:27
        throw new Error('Unrecognized cluster');
        ^

Error: Unrecognized cluster
    at new FuncPayload (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/functional.js:27:15)
    at /opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:33:22
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:121:9)
    at Object.onceWrapper (events.js:286:20)
    at Dissolve.emit (events.js:198:13)
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/dissolve-chunks/index.js:73:29)
    at Dissolve.emit (events.js:198:13)
    at emitReadable_ (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:456:10)
    at emitReadable (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:450:7)
    at readableAddChunk (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_readable.js:206:11)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-08-13T09_27_09_906Z-debug.log

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: "version":"1.4.0"
CC253X firmware version: "coordinator":20190608

zhapwey, 2019-08-13:
Between error and error I receive these messages:

  zigbee2mqtt:info 8/13/2019, 10:41:43 AM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 8/13/2019, 10:41:43 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"927c4db","coordinator":20190608,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 8/13/2019, 10:41:43 AM Mounted the cieApp (epId 11)
  zigbee2mqtt:debug 8/13/2019, 10:41:53 AM Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6ffffe51fffa"' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1
  zigbee2mqtt:debug 8/13/2019, 10:41:54 AM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"_TYST11_2dpplnsn","modelId":"dpplnsn\u0000","powerSource":3}}' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1
  zigbee2mqtt:debug 8/13/2019, 10:42:24 AM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"20180727"}}' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1
  zigbee2mqtt:debug 8/13/2019, 10:42:30 AM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe51fffa"'
  zigbee2mqtt:debug 8/13/2019, 10:42:33 AM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe51fffa"'
  zigbee2mqtt:debug 8/13/2019, 10:42:34 AM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe51fffa"'
  zigbee2mqtt:debug 8/13/2019, 10:42:34 AM Accept device incoming with ieeeAddr '0x000d6ffffe51fffa' permit join is 'true'
  zigbee2mqtt:debug 8/13/2019, 10:42:34 AM Allowing device '0x000d6ffffe51fffa' to join
  zigbee2mqtt:debug 8/13/2019, 10:42:34 AM Received zigbee message of type 'devIncoming' with data '"0x000d6ffffe51fffa"' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1
  zigbee2mqtt:info 8/13/2019, 10:42:34 AM Device incoming...
  zigbee2mqtt:info 8/13/2019, 10:42:34 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 8/13/2019, 10:42:34 AM Received zigbee message of type 'devStatus' with data '"online"' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1
  zigbee2mqtt:info 8/13/2019, 10:45:02 AM Succesfully reenabled joining
  zigbee2mqtt:debug 8/13/2019, 10:46:13 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"appVersion":255}}' of device 'undefined' (0x00124b0018ed1d2c) of endpoint 11
  zigbee2mqtt:debug 8/13/2019, 10:46:13 AM Ignoring message from coordinator
  zigbee2mqtt:debug 8/13/2019, 10:46:13 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"appVersion":255}}' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1
  zigbee2mqtt:debug 8/13/2019, 10:46:13 AM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"appVersion":255}}' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1
  zigbee2mqtt:warn 8/13/2019, 10:46:13 AM No converter available for '"_TYST11_2dpplnsn' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"appVersion":255}}'
  zigbee2mqtt:warn 8/13/2019, 10:46:13 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 8/13/2019, 10:46:14 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"appVersion":72}}' of device 'undefined' (0x00124b0018ed1d2c) of endpoint 11
  zigbee2mqtt:debug 8/13/2019, 10:46:14 AM Ignoring message from coordinator
  zigbee2mqtt:debug 8/13/2019, 10:46:14 AM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"appVersion":72}}' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1
  zigbee2mqtt:debug 8/13/2019, 10:46:14 AM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"appVersion":72}}' of device 'dpplnsn' (0x000d6ffffe51fffa) of endpoint 1
  zigbee2mqtt:warn 8/13/2019, 10:46:14 AM No converter available for '"_TYST11_2dpplnsn' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"appVersion":72}}'
  zigbee2mqtt:warn 8/13/2019, 10:46:14 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 8/13/2019, 10:46:39 AM Saving state to file /opt/zigbee2mqtt/data/state.json

I'm going to try the messages that need a converter, create the converters to see the messages ( zigbee2mqtt:warn 8/13/2019, 10:46:13 AM No converter available for '"_TYST11_2dpplnsn' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"appVersion":255}}'), but I don't know how to handle the messages (zigbee2mqtt:debug 8/13/2019, 10:42:30 AM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe51fffa"') or the error.

Thank you

zhapwey, 2019-08-14:
good:
reading several post and seen that to include new devices you must use the dev version.
I have changed to this version and receive two messages with problem detected.

I am not sure what to do with them.

PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
AssertionError [ERR_ASSERTION]: { frameCntl:
   { frameType: 1, manufSpec: 0, direction: 1, disDefaultRsp: 0 },
  manufCode: null,
  seqNum: 16,
  cmdId: 36 } deepEqual undefined
    at Object.header (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:90:20)
    at dispatchIncomingMsg (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:528:29)
    at Controller.incomingMsgHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:625:12)
    at Controller.emit (events.js:198:13)
    at Object.bridge._areqEventBridge (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/event_bridge.js:11:16)
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/controller.js:104:20)
    at Znp.emit (events.js:198:13)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/znp/znp.js:72:18)
    at Parser.emit (events.js:198:13)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/unpi/parser.js:39:26)
===== Arguments =====
<Buffer 09 10 24 00 00> 61184
===== Buffer =====
{ type: 'Buffer', data: [ 9, 16, 36, 0, 0 ] }
===== zcl-packet header =====
{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":0},"manufCode":null,"seqNum":16,"cmdId":36}
===== zcl header =====
undefined
===============

PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
AssertionError [ERR_ASSERTION]: { frameCntl:
   { frameType: 1, manufSpec: 0, direction: 1, disDefaultRsp: 0 },
  manufCode: null,
  seqNum: 13,
  cmdId: 2 } deepEqual undefined
    at Object.header (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:90:20)
    at dispatchIncomingMsg (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:528:29)
    at Controller.incomingMsgHandler (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/af.js:625:12)
    at Controller.emit (events.js:198:13)
    at Object.bridge._areqEventBridge (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/event_bridge.js:11:16)
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/lib/components/controller.js:104:20)
    at Znp.emit (events.js:198:13)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/znp/znp.js:72:18)
    at Parser.emit (events.js:198:13)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/unpi/parser.js:39:26)
===== Arguments =====
<Buffer 09 0d 02 00 05 6b 00 00 10 01 e0 00 dc 01 fe 00 dc 03 fc 00 dc 05 28 00 be> 61184
===== Buffer =====
{ type: 'Buffer',
  data:
   [ 9,
     13,
     2,
     0,
     5,
     107,
     0,
     0,
     16,
     1,
     224,
     0,
     220,
     1,
     254,
     0,
     220,
     3,
     252,
     0,
     220,
     5,
     40,
     0,
     190 ] }
===== zcl-packet header =====
{"frameCntl":{"frameType":1,"manufSpec":0,"direction":1,"disDefaultRsp":0},"manufCode":null,"seqNum":13,"cmdId":2}
===== zcl header =====
undefined
===============

If someone could tell me what to do with them I would appreciate it.

regards

Koenkk, 2019-08-15:
Are you sure this is with the latest dev? Because this issue should have been fixed.

zhapwey, 2019-08-16:
Hello:
Thanks for your interest.
I am using 1.5.1 and firmware 20190608.

I do not know how I can solve this, if you need any data or information do not hesitate to ask me.

Thank you

Koenkk, 2019-08-18:
https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

zhapwey, 2019-08-19:
good:
In the first message when I received an error in zigbee2mqtt of "Error: Unrecognized cluster", I used the normal version, but in the last message the version used is the dev.

hhalibo, 2019-09-07:
> good:
> In the first message when I received an error in zigbee2mqtt of "Error: Unrecognized cluster", I used the normal version, but in the last message the version used is the dev.

Did you have any progress?

zhapwey, 2019-09-11:
Good:
Thanks for your interest,
Install the new version of zigbee2mqtt 1.6 in development mode and now I only receive messages 'PM Device' 0x000d6ffffe51fffa 'messages announced itself'. And I don't know how to continue ...

Hacker-CB, 2020-09-24:
I got docs for tuya DP's for this thermostat from vendor.

[tuya DPs_translated.xlsx](https://github.com/Koenkk/zigbee2mqtt/files/5274121/tuya.DPs_translated.xlsx)

They told me, that next devices has same DPs:

- [HT-10] _TYST11_2dpplnsn (TRV)
- [HT-08] _TYST11_wv90ladg (Wall-mount thermostat) 

I will try to add it to z2m.

Hacker-CB, 2020-09-24:
- HT-08: http://www.etopcontrols.com/thermostat/detail/HT-08.html
- HT-10: http://www.etopcontrols.com/thermostat/detail/HT-10.html

Hacker-CB, 2020-09-24:
@Koenkk please reopen this ticket.
I hope i will finish support for this two devices in next days.

Hacker-CB, 2020-09-30:
All done. @Koenkk you can close issue

closed 2020-09-30 by Koenkk