Koenkk/zigbee2mqtt #1857: Converter for new cid

Hoanthieu, 2019-08-13:
Hi,
I try to connect a new smartlock to HA, and log show:
`No converter available for 'TY0A01' with cid '61184', type 'devChange' and data '{"cid":61184,"data":{"5":"TY0A01","6":"20181115","7":3,"65533":1}}'`
How can I write converter for that msg

ptvoinfo, 2019-08-14:
@Hoanthieu 
Docs: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Repository: https://github.com/Koenkk/zigbee-shepherd-converters/
There is a lot of converters for many devices. You may create a new converter based on these examples. Add a new converter for your reports one by one. Start with devChange.

closed 2019-10-20 by stale[bot]

angeck, 2020-05-07:
> Salut,
> j'essaie de connecter un nouveau smartlock à HA et de montrer le journal:
> `No converter available for 'TY0A01' with cid '61184', type 'devChange' and data '{"cid":61184,"data":{"5":"TY0A01","6":"20181115","7":3,"65533":1}}'`
> comment puis-je écrire un convertisseur pour ce msg

> Hi,
> I try to connect a new smartlock to HA, and log show:
> `No converter available for 'TY0A01' with cid '61184', type 'devChange' and data '{"cid":61184,"data":{"5":"TY0A01","6":"20181115","7":3,"65533":1}}'`
> How can I write converter for that msg

Bonjour,

Je suis nouveau dans zigbee2mqtt et je dispose du modèle TY0A01, qui n'est pas reconnu, avez-vous réussi l'inclusion ? Je suis beaucoup trop nul pour le tuto de zibgee2mqtt. Merci.

angeck, 2020-05-09:
Bonjour,

J'ai réussi :

zigbee2mqtt:info  2020-05-09 08:44:46: Successfully interviewed '0xec1bbdfffe781b28', device has successfully been paired
zigbee2mqtt:info  2020-05-09 08:44:46: Device '0xec1bbdfffe781b28' is supported, identified as: VIMA VIMA LOCK  (A6121)
zigbee2mqtt:info  2020-05-09 08:44:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe781b28","model":"A6121","vendor":"VIMA","description":"VIMA LOCK ","supported":true}}'

Bon par par contre pour le moment je n'arrive pas à en tirer quoi que ce soit. Même avec le conteneur du même modèle en Xiaomi.

szaman-89, 2020-10-12:
> Bonjour,
>
> J'ai réussi :
>
> zigbee2mqtt:info 2020-05-09 08:44:46: Successfully interviewed '0xec1bbdfffe781b28', device has successfully been paired
> zigbee2mqtt:info 2020-05-09 08:44:46: Device [...]

Did you managed to integrate this lock with z2m?