Koenkk/zigbee2mqtt #1858: Update installation documemntation for Pi Zero

PianSom, 2019-08-13:
# Bug Report
Documemtation is out of date
On 
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html
in section 2 it says to install node on a pi Zero like this 
````
# NOTE: For Raspberry Pi Zero follow https://warlord0blog.wordpress.com/2018/06/27/node-js-v8-on-raspberry-pi-zero/
````

While the link is still working, it refers to installation of node v8.11.3, which is no longer sufficiently current for zigbee2mqtt

I followed the same instructions but went for node v10.9.0 using 
````
wget https://nodejs.org/dist/v10.9.0/node-v10.9.0-linux-armv6l.tar.xz
tar -xvf node-v10.9.0-linux-armv6l.tar.xz
cd node-v10.9.0-linux-armv6l.tar
sudo cp -R * /usr/local/
````

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2019-08-14:
Could you make a pr to update https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/getting_started/running_zigbee2mqtt.md?

Koenkk, 2019-10-13:
Manual is updated

closed 2019-10-13 by Koenkk