Koenkk/zigbee2mqtt #1861: How to use on Android ( TV Boxes, phones via USB-OTG )

podarok, 2019-08-14:
Hi
I've managed to run this on android TV box. Down below is a recipe

1. Install SuperSU(Or any other su tool via recovery ) . This step is needed because /dev/ttyACM0 is not accessible for users under Android yet. See https://github.com/felHR85/UsbSerial/issues/272 
2. patch serialport with this https://github.com/podarok/node-serialport/commit/0f5cfd731545906d2b26f31d093d7c24acaf9d9a
3. run termux terminal https://play.google.com/store/apps/details?id=com.termux
4. chmod a+w /dev/ttyACM0
5. exit
6. cd zigbee2mqtt && yarn start
Proof of concept
[image]

Once step 1 is resolved this will open zigbee2mqtt for way more bigger Android community

Koenkk, 2019-08-15:
This is awesome! If https://github.com/serialport/node-serialport/pull/1912 is released and zigbee2mqtt is updated with it I think we should add this to the zigbee2mqtt docs.

reef-actor, 2019-09-18:
[serialport 8.0.1](https://www.npmjs.com/package/serialport/v/8.0.1) has recently been uploaded

Koenkk, 2019-09-18:
zigbee2mqtt dev branch uses serialport 8.0.1 know, assuming this can be closed. Thanks @podarok !

closed 2019-09-18 by Koenkk

data-ThermaLog, 2022-01-06:
@podarok  I'm trying to run zigbee2mqtt on my cell phone, but "npm ci" gives errors. Can you please help?

blakadder, 2022-09-17:
this might become more actual again because NSPanel Pro has a Zigbee chip and ADB/root access