Koenkk/zigbee2mqtt #1862: Two CC2531 sticks nearby = troubles

u17194907425, 2019-08-14:
Hi! I want to share an observation that potentially might be something more like a bug. Or (hopefully) just my lack of configuration experience.

I have a CC2531 in use for quite a while now. Sockets, lamps, switches from Xiaomi and IKEA. All good. A couple of weeks ago I started to prepare a setup for my parents: another PC with another CC2531 just located inside my house until I'm done with configuring and move it to their location.

After doing an initial installation I have found out that two of my Zigbee Sockets are reported as not available on my home setup. Hum... enabled pairing on my home's setup, re-paired both sockets. All good.

The very next day I saw the same happened to another socket. All that time the setup I was preparing for my parents was switched off, but the CC2531 stick was plugged and with power (green light was on). Hum... I have pulled that CC2531 off from the parent's PC and re-paired the socket back to my home CC2531. All good. 

Now today I was continuing with setting up the configs for my parents and again the issue. Now a Aquara bulb.

---

Now the details.

* The issue is (no route to the device) or can't ping the device.
* I'm using Home Assistant and a plugin `You are currently running version 1.5.1.1` (`zigbee2mqtt version 1.5.1 (commit #unknown)` from log)
* Both sticks are flashed with `Coordinator firmware version: '20190608'`
* The configs of the plugin on my systems and on parent's system are in `network_key` and in `pan_id`

---

Is that me who did something wrong? If so then what? I wonder if one day my neighbor decides to set up a smart home... will I see this issue every day? Will all of us see this?

ryanbeaton, 2019-08-15:
Hi chance have you set up both with defaults? They would both have the same channel, pan ID, extended pan ID and network encryption key if you did and would probably be confused.

https://www.zigbee2mqtt.io/configuration/configuration.html has more info

u17194907425, 2019-08-15:
Hey! @ryanbeaton. Thanks for a quick reply! Indeed they have the same `channel` (11 the default one). And I haven't configured the `ext_pan_id` so it has to be the same (default) as well.

But as I mentioned their `network_key` and `pan_id` are different. My expectation was - it is enough to separate the networks.

Do you think it is absolutely a must to change `ext_pan_id` and `channel` in order to avoid the described behavior? If so then I think there should be a warning about that. Smart Homes are becoming a huge trend and if networks are so easily influencing each other, then probably the default config is not the way to go in a long run.

closed 2019-10-21 by stale[bot]