Koenkk/zigbee2mqtt #1863: ZNMS12LM - Xiaomi Aqara S2 Lock (Router)

Omerdagan, 2019-08-15:
I've been having some problems with the lock, the lock/unlock state of the sensor within home assistant does not change at all, and the user id report when opening the lock isn't always working for some reason
MQTT log - https://hastebin.com/gatubahusi.apache

## Debug Info
zigbee2mqtt version: 1.5.1 
CC253X firmware version: 20190608

dzungpv, 2019-08-25:
It is normal, I am the author of the code support it, I use single message to send all it status, example when you open the lock form inside the is no user id and it will be null. And it only store last status in the mqtt message. when you from outside, first you touch the sensor, and then you open it, open is the last message without user id.
To store and display the last user you must use automation in home assistant:

```
  - id: notify_lock_open
    alias: "Notify when someone open lock"
    trigger:
      platform: template
      value_template: "{% if (state_attr('sensor.lock_id', 'user')!=None) %}true{% endif %}"
    condition:
      condition: template
      value_template: "{% if ((state_attr('sensor.lock_id', 'user')|int) > 0) %}true{% endif %}"
    action: 
      - service: mqtt.publish
        data_template:
          topic: "zigbee2mqtt/lock_id/last_open_user"
          payload_template: "{% set user_id = (state_attr('sensor.0x00158d000273ab71_action', 'user')|int) -%}
user_id 
                 {%- endif %}"     
          retain: true                 
```
and user mqtt sensor to display it.

My be i will add a field: last_user to the code but not now.

Omerdagan, 2019-08-25:
Thank you for the reply. My intention was that the lock status itself isn't changing, even if the door is left unlocked. Any way around it? Maybe in a similar way to the sensor automation you have included?

closed 2019-10-31 by stale[bot]