Koenkk/zigbee2mqtt #1864: Crashes with 'Ccznp has not been initialized yet'

riekelt, 2019-08-16:
# Bug Report
## What happened
Crashes with error `Error: ccznp has not been initialized yet` on starting zigbee2mqtt

## What did you expect to happen
Zigbee2mqtt to boot properly and start the network.

## How to reproduce it (minimal and precise)
Unknown

I've rebuild the whole network from scratch. Removed the data folder, reflashed the stick en walked around the house to re-pair everything. Network was stable after re-pairing everything, but failed with the same error message again after restarting.

## Debug Info
Number of devices: 15
zigbee2mqtt version: 1.5.1
CC253X firmware version:
I've tried both the source routing (20190619) and default (20190608) firmware on 2 different cc2531 sticks, but both give the same error.

[image]

## Crash log
https://pastebin.com/6K8k798g

Koenkk, 2019-08-18:
Could you try with: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing (doesn't need repairing).

riekelt, 2019-08-18:
Already tried that one (as mentioned in the issue). So, unfortunately, no luck with that firmware.

Koenkk, 2019-08-18:
Do you use the reporting or availability feature?

riekelt, 2019-08-18:
`homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://10.1.0.10'
  user: zigbee
  password: 'redacted'
#  include_device_information: true
serial:
  port: /dev/ttyZIGBEE0
advanced:
  report: true
  channel: 20
  log_level: debug
  network_key: 'redacted'
`

This is my config. But i'm not sure if the report: true was already there before the error first occurred. I've altered some of the config params along the way to debug the whole thing.

Have moved the stick to another PC  (from raspberry pi 3b+ with raspbian to a mac mini vmware esx server with centos) and the errors where gone. It's running stable now, but that might as well have something to do with the different location in the house and less devices being connected directly to the stick.

Will try with reporting: false on the raspberry again to see if that improves the stability.

Koenkk, 2019-08-19:
Can you provide the log when running with `DEBUG=* npm start`?

closed 2019-10-25 by stale[bot]