Koenkk/zigbee2mqtt #1865: Upgrade lightbulb firmware using zigbee2mqtt

foxy82, 2019-08-16:
From looking at deConz github it looks like they have managed to perform a light bulb firmware upgrade using their USB stick and not the bulbs proprietary hub.

Is this something that could be implemented into zigbee2mqtt? I'm particularly interested in the tradfri side.

Some links:

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/firmware_update.cpp

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/96

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1516

To get the IKEA OTA files:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/ikea-ota-download.py

Koenkk, 2019-08-18:
Note to self: functionality should be implemented in https://github.com/koenkk/zigbee-herdsman, zigbee2mqtt only should act like a frontend.

sjorge, 2019-10-08:
Looks like the deconz people also found a few Hue firmwares, although they are encrypted... apparently it doesn't matter as the client decodes it on device.

Is this something that could be worked on once zigbee-herdsman is stable?

Tosko4, 2019-11-17:
Hello friends! Has anyone had time and luck to develop this?

sjorge, 2019-11-17:
Based on Koenkk/zigbee-herdsman#60 @tb-killa was working on it in his own branch, last update to that was over a month ago so I don't think a whole lot of progress has been made.

kiall, 2019-12-07:
Also worth mentioning here - https://gist.github.com/hhromic/ddadf70f80aae7b861e4d593ac84cf52

this shows theres also a test update channel for IKEA

Koenkk, 2020-02-03:
Continuing here: https://github.com/Koenkk/zigbee-herdsman/issues/60

closed 2020-02-03 by Koenkk