Koenkk/zigbee2mqtt #1869: Osram A60 TW going offline

hipsi, 2019-08-17:
I have 3 devices connected in Home assistant. 2x WeMo bulbs and 1 Osram Lightify A60 TW. The Wemos work fine, but finding that the Osram keeps going offline.  If I power it off manually and back on again I'm able to control it, but at some point it stops being contactable.

I get the following error in this case: 
zigbee2mqtt:error 8/17/2019, 10:03:15 AM Zigbee publish to device '0x841826000000df7b', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 183. APS no ack.

## Debug Info
zigbee2mqtt version: 1.5.1.2
CC253X firmware version: 20190608

thomasBrecho, 2019-08-30:
Same Problem with my Configuration. Also Osram smart+ bulb E14 candle

0rn0lf, 2019-09-13:
Same here with my A60 W. At some point it looses the connection but comes back on immediately.
The problem is that the light turns on by default and theres no way to change this. You can imagine that its quite annoying when youre sleeping and the light turns on in the middle of the night.

The light also turns on when you restart zigbee2mqtt.

Koenkk, 2019-09-14:
@0rn0lf 

> The light also turns on when you restart zigbee2mqtt.

This is interesting, can you sniff the zigbee traffic when this happen? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

0rn0lf, 2019-09-15:
@Koenkk 
> This is interesting, can you sniff the zigbee traffic when this happen? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

Sorry, I cant download zboss. I dont receive the activation mail for my account. I tried 2 different mail providers

0rn0lf, 2019-09-26:
Still not possible to create a account on dsr-wireless.com.
Can anyone help me out with the 'ZBOSS Sniffer for Windows 64-bit' package?

0rn0lf, 2019-09-26:
@Koenkk 
> This is interesting, can you sniff the zigbee traffic when this happen? (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

Finally I found the time to do this. Attached you can find the wireshark file. Its just a restart of zigbee2mqtt and the light turns on automatically.
I hope you can find something. Do you need anything else?

[sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/3656674/sniff.zip)

Koenkk, 2019-09-26:
I indeed see an On command being executed, can you provide the logging when running with debug logging enabled?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

0rn0lf, 2019-09-26:
@Koenkk 
[debug_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3656841/debug_log.txt)

The blub is 0x7cb03eaa00ac34e2

Koenkk, 2019-09-27:
The bulb is turned on by an MQTT message:

```
  zigbee2mqtt:debug 9/26/2019, 10:44:37 AM Received MQTT message on 'zigbee2mqtt/0x7cb03eaa00ac34e2/set' with data '{"state": "ON"}'
```

You should find out who send this message, my guess is that it's either a retained MQTT message or the message is send by home assistant itself. Nevertheless it doesn't look like a zigbee2mqtt issue.

closed 2019-12-03 by stale[bot]