Koenkk/zigbee2mqtt #1870: hs1rc-m not able to detect action/buttons

accelle17, 2019-08-17:
Can't get to work Heiman HS1RC-M smart remote controller. THe device is able to pair and auto discovery produces 3 sensors (action, link quality and battery). I tried this with two remotes and same situation. I don't see any mqtt messages nor logs of the action.

Here's the log of the join, I tried clicking/long press/2clicking the device afterwards:
 
8/17/2019, 11:21:26 AM - info: Zigbee: allowing new devices to join.
8/17/2019, 11:21:26 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"unknown","coordinator":20190608,"log_level":"debug","permit_join":true}'
8/17/2019, 11:21:37 AM - debug: Received zigbee message of type 'devLeaving' with data '"0x005043c9e721332f"' of endpoint undefined
8/17/2019, 11:21:37 AM - warn: Message without device!
8/17/2019, 11:21:53 AM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721332f"'
8/17/2019, 11:21:53 AM - info: Connecting with device...
8/17/2019, 11:21:53 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
8/17/2019, 11:21:53 AM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721332f"'
8/17/2019, 11:21:53 AM - info: Connecting with device...
8/17/2019, 11:21:53 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
8/17/2019, 11:21:54 AM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721332f"'
8/17/2019, 11:21:54 AM - info: Connecting with device...
8/17/2019, 11:21:54 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
8/17/2019, 11:21:55 AM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721332f"'
8/17/2019, 11:21:55 AM - info: Connecting with device...
8/17/2019, 11:21:55 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
8/17/2019, 11:21:55 AM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721332f"'
8/17/2019, 11:21:55 AM - info: Connecting with device...
8/17/2019, 11:21:55 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
8/17/2019, 11:21:56 AM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721332f"'
8/17/2019, 11:21:56 AM - info: Connecting with device...
8/17/2019, 11:21:56 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
8/17/2019, 11:21:56 AM - debug: Accept device incoming with ieeeAddr '0x005043c9e721332f' permit join is 'true'
8/17/2019, 11:21:56 AM - debug: Allowing device '0x005043c9e721332f' to join
8/17/2019, 11:21:56 AM - debug: Received zigbee message of type 'devIncoming' with data '"0x005043c9e721332f"' of device 'undefined' (0x005043c9e721332f) of endpoint 1
8/17/2019, 11:21:56 AM - info: Device incoming...
8/17/2019, 11:21:56 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
8/17/2019, 11:21:56 AM - info: New device 'undefined' with address 0x005043c9e721332f connected!
8/17/2019, 11:21:56 AM - debug: Saving state to file /share/zigbee2mqtt/state.json
8/17/2019, 11:21:56 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x005043c9e721332f","meta":{}}'
8/17/2019, 11:21:56 AM - warn: Device with modelID 'undefined' is not supported.
8/17/2019, 11:21:56 AM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
8/17/2019, 11:21:56 AM - error: Zigbee cannot determine endpoint for '0x005043c9e721332f'
8/17/2019, 11:21:56 AM - error:  TypeError: Cannot set property 'onZclFoundation' of null
    at device.epList.forEach (/zigbee2mqtt-1.5.1/lib/extension/responder.js:25:36)
    at Array.forEach (<anonymous>)
    at ExtensionResponder.setupOnZclFoundation (/zigbee2mqtt-1.5.1/lib/extension/responder.js:23:27)
    at ExtensionResponder.onZigbeeMessage (/zigbee2mqtt-1.5.1/lib/extension/responder.js:35:14)
    at extensions.filter.forEach (/zigbee2mqtt-1.5.1/lib/controller.js:143:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/zigbee2mqtt-1.5.1/lib/controller.js:143:14)
    at Zigbee.onMessage (/zigbee2mqtt-1.5.1/lib/zigbee.js:275:18)
    at ZShepherd.emit (events.js:198:13)
    at ZShepherd.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/lib/shepherd.js:68:14)
8/17/2019, 11:22:44 AM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"Heiman","modelId":"RC_V14","powerSource":3}}' of device 'RC_V14' (0x005043c9e721332f) of endpoint 1
8/17/2019, 11:22:44 AM - debug: Configuring 0x005043c9e721332f (0x005043c9e721332f) ...
8/17/2019, 11:22:44 AM - info: MQTT publish: topic 'homeassistant/sensor/0x005043c9e721332f/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x005043c9e721332f","json_attributes_topic":"zigbee2mqtt/0x005043c9e721332f","name":"0x005043c9e721332f_action","unique_id":"0x005043c9e721332f_action_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c9e721332f"],"name":"0x005043c9e721332f","sw_version":"Zigbee2mqtt 1.5.1","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
8/17/2019, 11:22:44 AM - info: MQTT publish: topic 'homeassistant/sensor/0x005043c9e721332f/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x005043c9e721332f","json_attributes_topic":"zigbee2mqtt/0x005043c9e721332f","name":"0x005043c9e721332f_battery","unique_id":"0x005043c9e721332f_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c9e721332f"],"name":"0x005043c9e721332f","sw_version":"Zigbee2mqtt 1.5.1","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
8/17/2019, 11:22:44 AM - info: MQTT publish: topic 'homeassistant/sensor/0x005043c9e721332f/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x005043c9e721332f","json_attributes_topic":"zigbee2mqtt/0x005043c9e721332f","name":"0x005043c9e721332f_linkquality","unique_id":"0x005043c9e721332f_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c9e721332f"],"name":"0x005043c9e721332f","sw_version":"Zigbee2mqtt 1.5.1","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
8/17/2019, 11:22:47 AM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"20150427"}}' of device 'RC_V14' (0x005043c9e721332f) of endpoint 1
8/17/2019, 11:22:47 AM - warn: Failed to configure 0x005043c9e721332f (0x005043c9e721332f) ('TypeError: Cannot read property 'status' of undefined') (attempt #1)
8/17/2019, 11:22:47 AM - warn: This can be ignored if the device is working properly
8/17/2019, 11:22:51 AM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman"}}' of device 'RC_V14' (0x005043c9e721332f) of endpoint 1
8/17/2019, 11:22:51 AM - debug: Configuring 0x005043c9e721332f (0x005043c9e721332f) ...
8/17/2019, 11:22:53 AM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"RC_V14","dateCode":"20150427","powerSource":3,"locationDesc":"","physicalEnv":0}}' of device 'RC_V14' (0x005043c9e721332f) of endpoint 1
8/17/2019, 11:22:54 AM - warn: Failed to configure 0x005043c9e721332f (0x005043c9e721332f) ('TypeError: Cannot read property 'status' of undefined') (attempt #2)
8/17/2019, 11:22:54 AM - warn: This can be ignored if the device is working properly
8/17/2019, 11:22:57 AM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"deviceEnabled":1,"alarmMask":0}}' of device 'RC_V14' (0x005043c9e721332f) of endpoint 1
8/17/2019, 11:22:57 AM - debug: Configuring 0x005043c9e721332f (0x005043c9e721332f) ...
[... cut ...]
8/17/2019, 11:23:33 AM - debug: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'false'
8/17/2019, 11:23:33 AM - info: Zigbee: disabling joining new devices.
8/17/2019, 11:23:33 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"unknown","coordinator":20190608,"log_level":"debug","permit_join":false}'
8/17/2019, 11:23:37 AM - debug: Saving state to file /share/zigbee2mqtt/state.json

## Debug Info
zigbee2mqtt version: 1.5.1.2 hassio addon
CC253X firmware version: 20190608

Koenkk, 2019-08-18:
After pairing the device, make sure you see a `Successfully configured [DEVICE_IEEEADDR]`. You might need to keep it awake after pairing (by pressing some buttons on it).

accelle17, 2019-08-19:
i dont see that message, only "successfully reenabled joining" after the discovery

```
8/19/2019, 6:20:41 PM - debug: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
8/19/2019, 6:20:41 PM - info: Zigbee: allowing new devices to join.
8/19/2019, 6:20:41 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"unknown","coordinator":20190608,"log_level":"debug","permit_join":true}'
8/19/2019, 6:21:38 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"Heiman","modelId":"RC_V14","powerSource":3}}' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:21:38 PM - info: New device 'RC_V14' with address 0x005043c9e721312e connected!
8/19/2019, 6:21:38 PM - debug: Saving state to file /share/zigbee2mqtt/state.json
8/19/2019, 6:21:38 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x005043c9e721312e","meta":{"modelID":"RC_V14"}}'
8/19/2019, 6:22:08 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"20150427"}}' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:22:10 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":1,"manufacturerName":"Heiman"}}' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:22:14 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"RC_V14","dateCode":"20150427","powerSource":3,"locationDesc":"","physicalEnv":0}}' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:22:16 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"deviceEnabled":1,"alarmMask":0}}' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:22:16 PM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721312e"'
8/19/2019, 6:22:18 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"deviceEnabled":1,"alarmMask":0}}' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:22:46 PM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721312e"'
8/19/2019, 6:22:49 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"identifyTime":0}}' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:22:49 PM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721312e"'
8/19/2019, 6:22:53 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genAlarms","data":{"alarmCount":0}}' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:22:53 PM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721312e"'
8/19/2019, 6:22:59 PM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721312e"'
8/19/2019, 6:23:00 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genAlarms","data":{"alarmCount":0}}' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:23:01 PM - debug: Received zigbee message of type 'devInterview' with data '"0x005043c9e721312e"'
8/19/2019, 6:23:01 PM - debug: Accept device incoming with ieeeAddr '0x005043c9e721312e' permit join is 'true'
8/19/2019, 6:23:01 PM - debug: Allowing device '0x005043c9e721312e' to join
8/19/2019, 6:23:01 PM - debug: Received zigbee message of type 'devIncoming' with data '"0x005043c9e721312e"' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:23:01 PM - info: Device incoming...
8/19/2019, 6:23:01 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
8/19/2019, 6:23:01 PM - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:23:11 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x005043c9e721312e"' of device 'RC_V14' (0x005043c9e721312e) of endpoint 1
8/19/2019, 6:23:21 PM - info: Successfully reenabled joining
8/19/2019, 6:24:25 PM - debug: Saving state to file /share/zigbee2mqtt/state.json
```

kijuz, 2019-08-20:
i. i have the same problem with device rc-em version, model hs1rc-e.
battery and link quality are visible but nothing happens when push buttons.
i use the same device.js configuration of rc_v14 changing only device and model and adding this string to opt/zigbee2mqtt/lib/extension/homeassistant.js
663 'HS1RC-E': [configurations.sensor_action, configurations.sensor_battery]

ascglennzarb, 2019-08-30:
I am having a similar issue with the HS1RC-M.

I am running the Zigbee2Mqtt Addon (v1.5.1.2) on HomeAssistant Hass.io.

When a try to pair the device by clicking the reset button for 5 seconds, I get the below logs:

  zigbee2mqtt:info 8/30/2019, 11:07:50 AM Successfully reenabled joining
  zigbee2mqtt:info 8/30/2019, 11:08:02 AM New device 'RC_V14' with address 0x005043c953279572 connected!
  zigbee2mqtt:info 8/30/2019, 11:08:02 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x005043c953279572","meta":{"modelID":"RC_V14"}}'
  zigbee2mqtt:info 8/30/2019, 11:08:02 AM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_action","unique_id":"0x005043c953279572_action_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.5.1","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 8/30/2019, 11:08:02 AM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_battery","unique_id":"0x005043c953279572_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.5.1","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 8/30/2019, 11:08:02 AM MQTT publish: topic 'homeassistant/sensor/0x005043c953279572/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x005043c953279572","json_attributes_topic":"zigbee2mqtt/0x005043c953279572","name":"0x005043c953279572_linkquality","unique_id":"0x005043c953279572_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x005043c953279572"],"name":"0x005043c953279572","sw_version":"Zigbee2mqtt 1.5.1","model":"Smart remote controller (HS1RC-M)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:warn 8/30/2019, 11:08:04 AM Failed to configure 0x005043c953279572 (0x005043c953279572) ('TypeError: Cannot read property 'status' of undefined') (attempt #1)
  zigbee2mqtt:warn 8/30/2019, 11:08:04 AM This can be ignored if the device is working properly
  zigbee2mqtt:warn 8/30/2019, 11:08:07 AM Failed to configure 0x005043c953279572 (0x005043c953279572) ('TypeError: Cannot read property 'status' of undefined') (attempt #2)
  zigbee2mqtt:warn 8/30/2019, 11:08:07 AM This can be ignored if the device is working properly
  zigbee2mqtt:warn 8/30/2019, 11:08:10 AM Failed to configure 0x005043c953279572 (0x005043c953279572) ('TypeError: Cannot read property 'status' of undefined') (attempt #3)
  zigbee2mqtt:warn 8/30/2019, 11:08:10 AM This can be ignored if the device is working properly
  zigbee2mqtt:info 8/30/2019, 11:08:12 AM MQTT publish: topic 'zigbee2mqtt/0x005043c953279572', payload '{"battery":44.5,"linkquality":0}'
  zigbee2mqtt:info 8/30/2019, 11:08:16 AM MQTT publish: topic 'zigbee2mqtt/0x005043c953279572', payload '{"battery":44.5,"linkquality":0,"battery_low":false}'
  zigbee2mqtt:info 8/30/2019, 11:08:24 AM Device incoming...
  zigbee2mqtt:info 8/30/2019, 11:08:24 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 8/30/2019, 11:10:30 AM Successfully reenabled joining

In Home Assistance I can see the battery value and link quality value but nothing happens when I click on the buttons of the remote.

Has someone been able to successfully pair this device?

closed 2019-11-05 by stale[bot]

VenkateshGSRao, 2019-11-25:
Same problem here HS1RC-M paired successfully but nothing happens on pressing buttons.