Koenkk/zigbee2mqtt #1871: Not supported YALE Lock spamming with too many messages.

alexsahka, 2019-08-17:
Hi, I added a new device Yale lock YRD210 model: YRD210-HA-605.
Everything so far ok, except 3+ update messages per second from the lock.
New batteries.  Where is the problem config, or hardware problem? 
LOCK / UNLOCK works but because too many MQTT messages sending a command to lock is not always stable received by the device. 
Here is the config.

{
    zigbeeModel: ['YRD210 PB DB'],
    model: 'YRD210-HA-605',
    vendor: 'Yale',
    description: 'Assure lock',
    supports: 'lock/unlock, battery',
    fromZigbee: [fz.generic_lock, fz.battery_200],
        toZigbee: [tz.generic_lock],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('closuresDoorLock', 'lockState', 0, repInterval.HOUR, 0, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, repInterval.MAX, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },

Koenkk, 2019-08-18:
You probably want to use:

```js
(cb) => device.report('closuresDoorLock', 'lockState', 0, repInterval.HOUR, 1, cb),
```

alexsahka, 2019-08-18:
Thank you, but tried this and still MQTT posting a lot of rapid messages.
Maybe I added a new device not properly?
My steps are: 
1. Paired new device.
2. Added new device to devices.js
after correction now:
{
    zigbeeModel: ['YRD210 PB DB'],
    model: 'YRD210-HA-605',
    vendor: 'Yale',
    description: 'Assure lock',
    supports: 'lock/unlock, battery',
    fromZigbee: [fz.generic_lock, fz.battery_200],
        toZigbee: [tz.generic_lock],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('closuresDoorLock', 'lockState', 0, repInterval.HOUR, 1, cb),
                (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, repInterval.MAX, 0, cb),
            ];

            execute(device, actions, callback);
        },
    },

I didn't touch any other files, the question is what I did not correct?

alexsahka, 2019-08-18:
After deleting this line:
(cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, repInterval.MAX, 0, cb),

lock stops "spamming", will continue testing to see if it is permanent.

alexsahka, 2019-08-19:
After adding a user to lock, everything goes south...
no response from lock and error from zigbee2mqtt:

zigbee2mqtt:error 8/19/2019, 3:15:15 AM Zigbee publish to device 'yale_lock', closuresDoorLock - lockDoor - {"pincodevalue":""} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 240. MAC transaction expired. 

Deleting the user from the lock, gives me the same result.
If someone can help me with this would be really nice, I have no idea what I need to do.

Koenkk, 2019-08-21:
I don't have this lock myself.

@matisaul @CanuckianOz  did you also experience this issue?

matisaul, 2019-08-21:
I always receive that error: `Error: AF data request fails, status code: 240. MAC transaction expired.` but the lock works without problem. About Battery message problem, in my lock (YMF40) I don't receive any message of battery status, so I really don't know how to help you.

Maybe try to remove your device on zigbee2mqtt, and delete that line: `(cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, repInterval.MAX, 0, cb)` . Then, remove the battery and zigbee module from the lock, then put it back again, and turn on the "allow devices to join", and put the lock in pair mode.

alexsahka, 2019-08-24:
Lock removed from zigbee2mqtt, added back.
Line deleted : (cb) => device.report('genPowerCfg', 'batteryPercentageRemaining', 0, repInterval.MAX, 0, cb)

Still the same problems.

closed 2019-10-31 by stale[bot]