Koenkk/zigbee2mqtt #1872: Hue dimmer brightness parameter enhancement

stefanroelofs, 2019-08-17:
I paired a Hue Dimmer without issues. The device gives me messages in the log like this:

`8/17/2019, 7:24:28 PM - info: MQTT publish: topic 'zigbee2mqtt/HueRemote_DamiansRoom', payload '{"battery":77,"linkquality":18,"counter":1,"brightness":255}'
`
The parameter brightness is an integer that goes from 0 to 255.
It changes with increments of 50 when you press the Dimmer / Brighter buttons (Up/Down) on the remote.

When you repeatedly press the Up button, the range goes like this:
0 - 50 - 100 - 150 - 200 - 250 - 255
And when you press the down button, the range is like this:
255 - 205 - 155 - 105 - 55 - 5 - 0

In my opinion these improvements are possible:

The dim-levels are not uniformly divided across the range. The last press only adds 5.
If you increment with 32 the range is a lot nicer:
0 - 32 - 64 - 96 - 128 - 160 - 192 - 224 - 256

You can dim the light all the way down to 0. This means 'Off' in HomeAssistant. It would be nice to limit the lower value to 1. This way when you dim a light, it will always stay on.

I don't know if this is possible at all, it might just be a limitation of the device. Thanks anyway!
# Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: 20190608

Koenkk, 2019-08-18:
@ugrug what is your opinion about this?

james-fry, 2019-08-20:
FWIW I support @stefanroelofs suggestion.

stefanroelofs, 2019-09-20:
@ugrug No update on this?

james-fry, 2019-09-20:
Im using node red to implement 32 step dimming.
The reason is that I also have one of the trust dimmers that have similar physical layout (plus buttons to select group, which allows six different sets if controls) but lack the built in support for brightness.

Here is part of my flow:

````
[{"id":"871af3aa.f2eef","type":"switch","z":"3da2c349.0ebdcc","name":"On, off, up, down","property":"payload.action","propertyType":"msg","rules":[{"t":"eq","v":"on","vt":"str"},{"t":"eq","v":"off","vt":"str"},{"t":"eq","v":"up-press","vt":"str"},{"t":"eq","v":"down-press","vt":"str"}],"checkall":"true","repair":true,"outputs":4,"x":870,"y":1260,"wires":[["4ee0c9c1.0fd358"],["f202f877.ff4ae8"],["78a74e7c.8701b"],["e8618271.79e67"]]},{"id":"78a74e7c.8701b","type":"api-current-state","z":"3da2c349.0ebdcc","name":"","server":"b8197728.8ac978","version":1,"outputs":1,"halt_if":"","halt_if_type":"str","halt_if_compare":"is","override_topic":false,"entity_id":"light.bedside_james","state_type":"str","state_location":"payload","override_payload":"msg","entity_location":"data","override_data":"msg","blockInputOverrides":false,"x":1300,"y":1300,"wires":[["f6063295.d2b4a"]]},{"id":"e8618271.79e67","type":"api-current-state","z":"3da2c349.0ebdcc","name":"","server":"b8197728.8ac978","version":1,"outputs":1,"halt_if":"","halt_if_type":"str","halt_if_compare":"is","override_topic":false,"entity_id":"light.bedside_james","state_type":"str","state_location":"payload","override_payload":"msg","entity_location":"data","override_data":"msg","blockInputOverrides":false,"x":1300,"y":1360,"wires":[["4030d204.dfa58c"]]},{"id":"f6063295.d2b4a","type":"function","z":"3da2c349.0ebdcc","name":"Set brightness in payload +32","func":"var brightness = msg.data.attributes.brightness;\n\nif (brightness < 32) {\n   brightness = 32;\n}  else if (brightness >= 32 && brightness < 64) {\n   brightness = 64;\n}  else if (brightness >= 64 && brightness < 96) {\n   brightness = 96;\n}  else if (brightness >= 96 && brightness < 128) {\n   brightness = 128;\n}  else if (brightness >= 128 && brightness < 160) {\n   brightness = 160;\n}  else if (brightness >= 160 && brightness < 192) {\n   brightness = 192;\n}  else if (brightness >= 192 && brightness < 224) {\n   brightness = 224;\n}  else if (brightness >= 224) {\n   brightness = 255;\n} \n\nmsg.payload = {\n domain: 'light',\n service: 'turn_on',\n data: { entity_id: 'light.bedside_james', brightness: brightness}\n};\nreturn msg;","outputs":1,"noerr":0,"x":1650,"y":1300,"wires":[["367d8265.bc17ee"]]},{"id":"4030d204.dfa58c","type":"function","z":"3da2c349.0ebdcc","name":"Set brightness in payload -32","func":"var brightness = msg.data.attributes.brightness;\n\nif (brightness > 224) {\n   brightness = 224;\n}  else if (brightness > 192 && brightness < 225) {\n   brightness = 192;\n}  else if (brightness > 160 && brightness < 193) {\n   brightness = 160;\n}  else if (brightness > 128 && brightness < 161) {\n   brightness = 128;\n}  else if (brightness > 96 && brightness < 129) {\n   brightness = 96;\n}  else if (brightness > 64 && brightness < 97) {\n   brightness = 64;\n}  else if (brightness > 32 && brightness < 65) {\n   brightness = 32;\n}  else if (brightness < 32) {\n   brightness = 0;\n}\n\nmsg.payload = {\n domain: 'light',\n service: 'turn_on',\n data: { entity_id: 'light.bedside_james', brightness: brightness}\n};\nreturn msg;\n","outputs":1,"noerr":0,"x":1640,"y":1360,"wires":[["367d8265.bc17ee"]]},{"id":"367d8265.bc17ee","type":"api-call-service","z":"3da2c349.0ebdcc","name":"Set James Bedside Brightness","server":"b8197728.8ac978","service_domain":"light","service":"turn_on","data":"{\"entity_id\":\"light.bedside_james\"}","mergecontext":"","output_location":"payload","output_location_type":"msg","x":1970,"y":1300,"wires":[[]]},{"id":"b8197728.8ac978","type":"server","z":"","name":"Home Assistant instance","legacy":false,"hassio":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true}]
````

james-fry, 2019-09-20:
BTW it is possible to change this.
I looked in the code and its in the [fromZigbee.js](https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js):

```
if (button && type) {
                if (type == 'press') {
                    store[deviceID].pressStart = Date.now();
                    store[deviceID].pressType = 'press';
                    if (brightnessSend) {
                        const newValue = store[deviceID].brightnessValue + (button === 'up' ? 50 : -50);
                        store[deviceID].brightnessValue = numberWithinRange(newValue, 0, 255);
                    }
```

So I am sure the change can be made....

stefanroelofs, 2019-09-20:
Nicely found!
So that's an easy fix then :)

Here you can also change the 0 to a 1 in the numberWithinRange method call.
To prevent switching the light off when dimming all the way down.

closed 2019-09-24 by stefanroelofs