Koenkk/zigbee2mqtt #1873: Devices net being reached by ping, should be marked as unavailable

gaggio, 2019-08-18:
I have bulbs that are normally operated by wall switch, therefore not always powered. While the log shows that these bulbs are not responding to pings, they are still marked as "on" in home assistant, which I believe is not correct. There should be a (configurable) number of unresponded pings that cause the device to be considered unavailable.

Koenkk, 2019-08-19:
Do you use the availability feature? (https://www.zigbee2mqtt.io/configuration/configuration.html) if so, please provide logs when device ping fails and it is still marked as available.

gaggio, 2019-08-22:
It seems like I completely failed reading properly the documentation!
Works as intended with the availability feature activated.

Cheers

closed 2019-08-22 by gaggio