Koenkk/zigbee2mqtt #1874: ${message.groupID} in lib/controller maybe wrong case.

sjorge, 2019-08-18:
I'm trying to get one of the IKEA 5 button remotes to work on dev but hitting a small snag:

```zigbee2mqtt:debug 8/18/2019, 1:32:18 PM Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x90fd9ffffee8e67a) with groupID undefined of endpoint 1```

So which makes no sense given the code here:
https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L136

As we check if we have a message.groupid beforehand. I then noticed that the variable called when printing has ID in upper case while the check in the tenary if has it in lowercase.

(I was poking around while on the train, will check if this is indeed the issue later today or tomorrow)

sjorge, 2019-08-18:
Nice! That is the problem, I asked someone to push a button on my remote at home.

```
  zigbee2mqtt:debug 8/18/2019, 3:58:47 PM Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x90fd9ffffee8e67a) with groupID 52963 of endpoint 1
```

Koenkk, 2019-08-19:
Merged #1875 assuming this is fixed, thanks!

closed 2019-08-19 by Koenkk