Koenkk/zigbee2mqtt #1876: Don’t allow group and device to have the same friendly_name

sjorge, 2019-08-18:
A group and a device can have the same name, and this will share a mqtt topic, this causes problems.

e.g. bedroom_remote as the friendly name for a teådfri remote, but I called the group the same and this broke some things until I figure out what went wrong.

This can probably be fixed in a few ways
- enduring a friendly_name is unique across all devices AND groups
- prefixing group and device topics with a keyword (this is cleaner but will probably break existing setups — but it could be made optional maybe?)

(e.g. zigbee2mqtt/device/<friendly_name> and zigbee2mqtt/group/<friendly_name>)

Koenkk, 2019-08-19:
Should definitely be fixed, will look into it.

carldebilly, 2019-08-23:
It would be a breaking change to move those topics :-(.

Anyway, if you plan to do this, I'll have to update my GUI tool :-).

sjorge, 2019-08-23:
Yeah, I personally feel option 2 is cleanest. But it also breaks things.
So that is why I listed option 1 first.

I guess doing both might work, and then have a toggle to disable the device/ group/ prefix. But that does seem like a whole lot more work. If I could figure out how to do it, i’d probably go that route. But I can only do some basic sruff, I find it hard to wrap my head around z2m / herdsmen seperation.

Koenkk, 2019-08-26:
*enduring a friendly_name is unique across all devices AND groups* seems like the best solution to me, will add it after 1.6 in dev.

sjorge, 2019-09-07:
Some other minor things I noticed, devices can have a '/' in the name. (e.g. 'bedroom/sensor')

This is handy to embed a room seperator in the mqtt topic, so it's easy to get all events for a room, this does not work for groups. When naming a group 'bedroom/lamps' I cannot add or remove other devices from it.

If I name the group 'bedroom_lamps' it works fine.

carldebilly, 2019-09-07:
I don't think that's a good idea to put a `/` in the name of the device.

davekr8, 2019-09-26:
Maybe the validity if zigbee conform names should be reflected, e.g avoiding + or # in names. The / is not an issue from my point of view, mainly adding just a subtopic as additional hierarchy.

Koenkk, 2019-09-27:
Implemented, `friendly_name` has to be unique now across groups and devices.

closed 2019-09-27 by Koenkk