Koenkk/zigbee2mqtt #1879: Add support Aqara curtain b1 (ZNCLDJ12LM)

freenetwork, 2019-08-20:
Control message work from ZNCLDJ11LM, but cannot get responce with current position

I capture packets for detect battery level, change direction, auto close/open, clear postion.
I cannot determine responce witch current position

https://github.com/freenetwork/ZNCLDJ12LM

Koenkk, 2019-08-21:
*I cannot determine responce witch current position* -> are you sure that the ZNCLDJ11LM does support this?

freenetwork, 2019-08-22:
I am not sure about that. Since I do not have a previous curtain model. I will capture messages the other day with a position.

Koenkk, 2019-08-22:
So it seems it isn't supported (https://github.com/Koenkk/zigbee-shepherd-converters/issues/545#issuecomment-523628117). That means that we can integrate it, can you make a PR?

freenetwork, 2019-08-25:
I catch packets. Control message work from old model. Packets in my repo

freenetwork, 2019-08-27:
Dresden Electronic was aded the aqara b1

Koenkk, 2019-10-17:
Supported in latest dev now.

closed 2019-10-17 by Koenkk

ArkadiuszNiemiec, 2019-11-19:
@Koenkk is that intended?
[image]

Koenkk, 2019-11-19:
Looks like a zigbee2mqtt assistant issue, discoveries for this device are in place (https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L608). Please open an issue there.

ArkadiuszNiemiec, 2019-11-19:
Oh ok, it merged back to 1 device after restart. Thank you.