Koenkk/zigbee2mqtt #1882: Possible to create "offline" message  in JSON format?

bartplessers, 2019-08-21:
# Bug Report
## What happened
When zigbee2mqtt comes offline, it sends a message to the mqtt broker:

```
object
topic: "zigbee2mqtt/bridge/state"
payload: "offline"
qos: 0
retain: false
_topic: "zigbee2mqtt/bridge/state"
_msgid: "bcbe09ab.595818"
```

The payload here is just a string.

## What did you expect to happen
It would be more convenient to get a JSON format here, i.e.
```
object
topic: "zigbee2mqtt/bridge/state"
payload: "{ "state" : offline" }"
qos: 0
retain: false
_topic: "zigbee2mqtt/bridge/state"
_msgid: "bcbe09ab.595818"
```

Reason for this: I capture all messages of topic zigbee2mqtt/# in node-red, and post those messages to my logging backend.
It would be easier if I can always expect JSON as payload

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2019-08-22:
The reason that this message is a plain string is because of Home Assistant (which most of the zigbee2mqtt users use) expects it to be.

sjorge, 2019-08-22:
You can just use a change node in node-red to change this.

```
[{"id":"7b384f0e.cf442","type":"change","z":"52e0cfe2.a67ae8","name":"","rules":[{"t":"move","p":"payload","pt":"msg","to":"payload.state","tot":"msg"},{"t":"change","p":"payload.state","pt":"msg","from":"online","fromt":"str","to":"true","tot":"bool"},{"t":"change","p":"payload.state","pt":"msg","from":"offline","fromt":"str","to":"false","tot":"bool"}],"action":"","property":"","from":"","to":"","reg":false,"x":480,"y":360,"wires":[[]]}]
```

However I prefer to do this in a function
```
switch (msg.topic) {
    case "zigbee2mqtt/bridge/state":
        msg.payload = {
            "state": (msg.payload === "online"),
        };
        break;
}
```

closed 2019-10-28 by stale[bot]