Koenkk/zigbee2mqtt #1883: double / duplicated messages?

bartplessers, 2019-08-21:
# Bug Report
## What happened
Raspberry + CC2531 + Zigbee2Mqtt + Xiaomi Aqara Temp sensor

In the logs I see

```
8/21/2019, 3:02:05 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7f2df', payload '{"temperature":21.92,"linkquality":73,"humidity":58.57,"pressure":1021,"battery":91,"voltage":2985}'
8/21/2019, 3:02:05 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7f2df', payload '{"temperature":21.92,"linkquality":73,"humidity":58.57,"pressure":1021,"battery":91,"voltage":2985}'

(...)

8/21/2019, 3:09:49 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7ed23', payload '{"temperature":31.51,"linkquality":26,"humidity":39.14,"pressure":1020,"battery":80,"voltage":2965}'
8/21/2019, 3:09:49 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002d7ed23', payload '{"temperature":31.51,"linkquality":26,"humidity":39.14,"pressure":1020,"battery":80,"voltage":2965}'

etc...
```

A lot of these messages contain the same info (combination of THP, battery,linkquality, etc... is the same) on the same time.

Is this an issue of my sensors (sending double information)?
Or is it an issue of zigbee2mqtt?

## What did you expect to happen
less messages

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: CC2531_DEFAULT_20190608.zip
XIAOMI temp sensor:
https://www.banggood.com/nl/Original-Aqara-Atmos-Version-Temperature-Humidity-Sensor-Smart-Home-Thermometer-Hygrometer-p-1148666.html

Koenkk, 2019-08-22:
Can you provide the debug logging of these messsages?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Kryzek, 2019-08-23:
Just to hijack thread, I also see these same double reports with Airam LED OP A60 ZB 9W/827 E27. Device in question reports state changes (and values within) twice, with first attReport and then devChange and both trigger dish out mqtt-messages:
```
  zigbee2mqtt:debug 8/23/2019, 8:21:03 AM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c) of endpoint 1
  zigbee2mqtt:info 8/23/2019, 8:21:03 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"state":"OFF","linkquality":15,"brightness":162}'
  zigbee2mqtt:debug 8/23/2019, 8:21:03 AM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'ZBT-DimmableLight' (0x00158d0001cded1c) of endpoint 1
  zigbee2mqtt:info 8/23/2019, 8:21:03 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001cded1c', payload '{"state":"OFF","linkquality":15,"brightness":162}'
```

Also Ikea Control Outlet (E1603/E1702) reports changes with attReport and devChange:
```
  zigbee2mqtt:debug 8/23/2019, 8:28:07 AM Received MQTT message on 'zigbee2mqtt/0x000b57fffeef4be3/set' with data 'OFF'
  zigbee2mqtt:info 8/23/2019, 8:28:07 AM Zigbee publish to device '0x000b57fffeef4be3', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - 1
  zigbee2mqtt:info 8/23/2019, 8:28:07 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffeef4be3', payload '{"state":"OFF","linkquality":2}'
  zigbee2mqtt:debug 8/23/2019, 8:28:07 AM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'TRADFRI control outlet' (0x000b57fffeef4be3) of endpoint 1
  zigbee2mqtt:info 8/23/2019, 8:28:07 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffeef4be3', payload '{"state":"OFF","linkquality":5}'
  zigbee2mqtt:debug 8/23/2019, 8:28:07 AM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'TRADFRI control outlet' (0x000b57fffeef4be3) of endpoint 1
  zigbee2mqtt:debug 8/23/2019, 8:28:11 AM Received MQTT message on 'zigbee2mqtt/0x000b57fffeef4be3/set' with data 'ON'
  zigbee2mqtt:info 8/23/2019, 8:28:11 AM Zigbee publish to device '0x000b57fffeef4be3', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - 1
  zigbee2mqtt:info 8/23/2019, 8:28:11 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffeef4be3', payload '{"state":"ON","linkquality":5}'
  zigbee2mqtt:debug 8/23/2019, 8:28:11 AM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0x000b57fffeef4be3) of endpoint 1
  zigbee2mqtt:info 8/23/2019, 8:28:11 AM MQTT publish: topic 'zigbee2mqtt/0x000b57fffeef4be3', payload '{"state":"ON","linkquality":0}'
  zigbee2mqtt:debug 8/23/2019, 8:28:11 AM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0x000b57fffeef4be3) of endpoint 1
```

Koenkk, 2019-08-23:
The devChange messages will be gone once zigbee-shepherd is replace by zigbee-herdsman (will probably land in the dev branch after zigbee2mqtt 1.6).

closed 2019-10-22 by Koenkk