Koenkk/zigbee2mqtt #1886: No converter available for 'E1746' with cid 'genIdentify'

carldebilly, 2019-08-23:
I'm seeing this warning in the output log:

```
warn: No converter available for 'E1746' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"identifyTime":0}}'
```
Looks like the E1746 (Ikea TRÅDFRI signal repeater) is reporting something unhandled by Zigbee2Mqtt.

AdamWhittingham, 2019-08-24:
I've literally just hit this as well, while trying to debug my issue mentioned in #1100.

I'll try following the steps to enrol a new device and see if it at least connects the booster again.

Koenkk, 2019-08-26:
`devChange` is removed in zigbee-herdsman, will be automatically fixed after #1888, for now this can be ignored (it's not a problem).

james-fry, 2019-08-27:
It’s off topic, but how do you guys find the Ikea routers? I’m having quite a few problems with devices on the edge of my network in a noisy 2.4ghz environment 

Considering changing from cc2531 sticks as routers to the Ikea signal repeaters...

austrisv, 2019-08-28:
@james-fry what are the problems/how do you measure/qualify/repeat them?
For me it's hard to comment at a good level, as I got mine installed only a day ago.  Few observations:
Trying to monitor temp, submitting AC failure tickets before people start complaining:) Need to get the signal through few floors in a office building (center of city plus a lot of business with their own routers - must be noisy 2.4ghz).  Have also ordered cc2531 and, while waiting, playing with the tradfri repeaters.
This morning one aqara temp sensor (routed through router) was not reporting for >90minutes or so (the temp was changing by ~10degrees as AC got switched on; and if not changing, it should report once an hour, right?) ... when clicking sensors button it got alive with linkquality114 and continued working nicely.
router2repeater shows linkquality68, repeater2router linkquality 29.

james-fry, 2019-08-28:
I started adding a few more routers and the situation is improved.
Changing channel is possible but very unappealing due to need to repair a lot of devices.

closed 2019-11-03 by stale[bot]