Koenkk/zigbee2mqtt #1891: Can't add custom device (SyntaxError: Unexpected token :) - Philips Hue LST003

echox, 2019-08-24:
# Bug Report

Can't add a custom device: Philips HUE LED Outdoor Lightstrip LST003.

## What happened

I tried following the "Adding a new device guide" but somehow the configuration file produces syntax errors during parsing:

```> node index.js
/zigbee2mqtt-1.5.1/node_modules/zigbee-shepherd-converters/devices.js:3
    model: '9290018187B',
         ^
SyntaxError: Unexpected token :
    at Module._compile (internal/modules/cjs/loader.js:721:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:690:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-shepherd-converters/index.js:3:17)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
npm
```

## What did you expect to happen

Get a rid of the error message about an unsupported device.

## How to reproduce it (minimal and precise)

- Added configuration parameter: `  "zigbee_shepherd_devices": true`
- Added `devices.js` under `/share/zigbee2mqtt` with the following content:
```
{
    zigbeeModel: ['LST003'],
    model: '9290018187B',
    vendor: 'Philips',
    description: 'Hue ambiance LightStrip',
    supports: 'll see ' ,
    fromZigbee: [], // We will add this later
    toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee, // Should be empty, unless device can be controlled (e.g. lights, switches)
},
```
(Source: https://github.com/Koenkk/zigbee2mqtt/issues/332#issuecomment-418648328)

Result is the error message above.

I tried several other devices.js examples (even tried to use a valid device from the repo) and entered it by typing to prevent any copy and paste errors or strange line endings messing up the parser.

Has something changed in the structure of the file or is it just a typo I can't figure out?

## Debug Info
zigbee2mqtt version: 1.5.1.2 
CC253X firmware version: (right now unknown but flashed the version ~ 3 weeks ago)

echox, 2019-08-24:
Ok, I have to add it to the full devices.js file, not only add an empty file.
It works now and I will create a pull request :)

closed 2019-08-24 by echox

robinjanssen91, 2019-09-26:
Hi @echox , I have the same problem on Hassio, could you elaborate what you mean with 'full devices.js', I also made an empty file since I couldn't find any devices.js in the folder. Thanks!

echox, 2019-09-26:
Hi @robinjanssen91 , sure. The devices.js inside the other docker container will be replaced in full by the devices.js you configure. So you need to get the complete file and add your new device.
In hassio its a little bit tricky, because you are inside a docker container if you connect via SSH.
The documentation was a little bit misleading but is already updated, see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html "Persisting a custom devices.js", Step 3.
You can find the file in the repo here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js
Be careful with the version (if you're not running latest).

robinjanssen91, 2019-09-26:
Thanks @echox ! Your instructions worked! I used the samba share plugin on Hassio to remotely view my files and pasted the full devices.js in share/zigbee2mqtt.

misteriks, 2020-02-25:
fyi, I had some troubles getting zigbee_shepherd_devices: true running with a copied devices.js because the latest zigbee2mqtt supports OTA which is referenced a lot in the original devices.js but not supportded by the hassio zigbee2mqtt version