Koenkk/zigbee2mqtt #1892: zigbee2mqtt won't start after upgrade to version 1.5.1: Cannot find module 'zigbee-herdsman/dist/ziee'

Aberixis7306, 2019-08-25:
Hello all!

After upgrading to version 1.5.1 starting zigbee2mqtt it **cannot find the module 'zigbee-herdsman/dist/ziee'** and the service stops. I'm using openHAB on a Raspberry Pi 3B+.

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: 20190608

Short debug listing:
```
systemd[1]: Started zigbee2mqtt.                                                                              
npm[22732]: > zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt                                                        
npm[22732]: > node index.js                                                                                   
npm[22732]: internal/modules/cjs/loader.js:638                                                                
npm[22732]:     throw err;                                                                                    
npm[22732]:     ^                                                                                             
npm[22732]: Error: Cannot find module 'zigbee-herdsman/dist/ziee'                                             
npm[22732]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)                   
npm[22732]:     at Function.Module._load (internal/modules/cjs/loader.js:562:25)                              
npm[22732]:     at Module.require (internal/modules/cjs/loader.js:692:17)                                     
npm[22732]:     at require (internal/modules/cjs/helpers.js:25:18)                                            
npm[22732]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)                                 
npm[22732]:     at Module._compile (internal/modules/cjs/loader.js:778:30)                                    
npm[22732]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)                      
npm[22732]:     at Module.load (internal/modules/cjs/loader.js:653:32)                                        
npm[22732]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)                                      
npm[22732]:     at Function.Module._load (internal/modules/cjs/loader.js:585:3)                               
npm[22732]: npm ERR! code ELIFECYCLE                                                                          
npm[22732]: npm ERR! errno 1                                                                                  
npm[22732]: npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`                                                 
npm[22732]: npm ERR! Exit status 1                                                                            
npm[22732]: npm ERR!                                                                                          
npm[22732]: npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.                                            
npm[22732]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm[22732]: npm ERR! A complete log of this run can be found in:                                              
npm[22732]: npm ERR!     /home/openhabian/.npm/_logs/2019-08-25T09_08_52_299Z-debug.log                       
systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE                           
systemd[1]: zigbee2mqtt.service: Unit entered failed state.                                                   
systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.                                              
systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.                              
systemd[1]: Stopped zigbee2mqtt.                  
```

Can somebody help me?

Elliott32224, 2019-08-25:
Same problem here and I have not been able to fix, yet.

Koenkk, 2019-08-26:
After updating you need to do a `rm -rf node_modules && npm install`

Aberixis7306, 2019-08-26:
I already did this before, finding your hint in another thread. But with no luck ...
Tried it once again - here ist the output of the npm installation procedure:
```
> @serialport/bindings@2.0.8 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.16.3 runtime=node arch=arm libc= platform=linux)
make: Verzeichnis „/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build“ wird betreten
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Verzeichnis „/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build“ wird verlassen

> @serialport/bindings@2.0.8 install /opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.16.3 runtime=node arch=arm libc= platform=linux)
make: Verzeichnis „/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build“ wird betreten
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Verzeichnis „/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build“ wird verlassen

> core-js-pure@3.1.4 postinstall /opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/core-js-pure
> node scripts/postinstall || echo "ignore"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 2064 packages from 606 contributors and audited 874609 packages in 264.645s
found 1 moderate severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details
```

tristone-cz, 2019-08-26:
Absolutely the same here.

Trying to install on Raspebrry Pi. Node version 10.16.3, Npm 6.9.0

I am trying to do fresh install, no update.

wossix, 2019-08-27:
Hey,

I did faced the same issue, yesterday. Today I did some further testing.
My solution was: **not** fixing the shown audit issues.

After doing a "npm audit fix" the "npm start" provides me with the same output as @Aberixis7306 provided earlier.

Skipping the audit; "npm start" is coming up with:

```
> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info 8/27/2019, 6:47:55 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-08-27.18-47-55'
  zigbee2mqtt:info 8/27/2019, 6:47:55 PM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
  zigbee2mqtt:info 8/27/2019, 6:47:55 PM Starting zigbee-shepherd
  zigbee2mqtt:info 8/27/2019, 6:47:55 PM Error while starting zigbee-shepherd

```
Last line is only because I didn't moved the stick to the new environment yet.

@Koenkk 

```
/opt/zigbee2mqtt-151# npm audit
=== npm audit security report ===                        
                                                                                
# Run  npm install zigbee-herdsman@0.6.0  to resolve 1 vulnerability
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Moderate      │ Memory Exposure                                              │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ bl                                                           │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ zigbee-herdsman                                              │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ zigbee-herdsman > dissolve > bl                              │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://npmjs.com/advisories/596                             │
└───────────────┴──────────────────────────────────────────────────────────────┘

# Run  npm update eslint-utils --depth 2  to resolve 1 vulnerability
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Critical      │ Arbitrary Code Execution                                     │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ eslint-utils                                                 │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ eslint [dev]                                                 │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ eslint > eslint-utils                                        │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ More info     │ https://npmjs.com/advisories/1118                            │
└───────────────┴──────────────────────────────────────────────────────────────┘

┌──────────────────────────────────────────────────────────────────────────────┐
│                                Manual Review                                 │
│            Some vulnerabilities require your attention to resolve            │
│                                                                              │
│         Visit https://go.npm.me/audit-guide for additional guidance          │
└──────────────────────────────────────────────────────────────────────────────┘
┌───────────────┬──────────────────────────────────────────────────────────────┐
│ Moderate      │ Memory Exposure                                              │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Package       │ bl                                                           │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Patched in    │ >=0.9.5 <1.0.0 || >=1.0.1                                    │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Dependency of │ zigbee-shepherd-converters                                   │
├───────────────┼──────────────────────────────────────────────────────────────┤
│ Path          │ zigbee-shepherd-converters > zigbee-herdsman > dissolve > bl │
├───────────────┼──────────────────────────────────────────────────────────────┤
[... 17 lines cut ...]
npm info lifecycle undefined~postshrinkwrap: undefined
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Valid OS:    darwin
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Valid Arch:  any
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Actual OS:   linux
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Actual Arch: x64
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Valid OS:    darwin
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Valid Arch:  any
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Actual OS:   linux
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Actual Arch: x64
[... cut ...] fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Valid OS:    darwin
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Valid Arch:  any
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Actual OS:   linux
npm verb notsup SKIPPING OPTIONAL DEPENDENCY: Actual Arch: x64

```

I hope I have been of help.

Cheers

Aberixis7306, 2019-08-27:
Not running "npm audit fix" unfortunately did not solve my problem. I'll try a clean install within the next few days.

What I also noticed is that the green LED on the CC2531 sniffer suddenly turns off about 1 minute after attaching the stick. I checked the serial port and the setting in "configuration.yaml" (/dev/ttyACM0), everything seems to be ok.

wossix, 2019-08-28:
Hey @Elliott32224, @Aberixis7306, you are right and sorry for late reply.

A  "rm -r node_modules & npm install" will not fix it.

Please try after the install: "npm install zigbee-herdsman@0.1.17". That is the version which gets installed with an initial installation of zigbee2mqtt_v.1.5.1.

@Koenkk 
```
# npm audit
                                                                                
                       === npm audit security report ===                        
                                                                                
# Run  npm install zigbee-herdsman@0.6.1  to resolve 1 vulnerability
                                                                                
  Moderate        Memory Exposure                                               
                                                                                
  Package         bl                                                            
                                                                                
  Dependency of   zigbee-herdsman                                               
                                                                                
  Path            zigbee-herdsman > dissolve > bl                               
                                                                                
  More info       https://npmjs.com/advisories/596                              
                                                                                

# Run  npm update eslint-utils --depth 2  to resolve 1 vulnerability
                                                                                
  Critical        Arbitrary Code Execution                                      
                                                                                
  Package         eslint-utils                                                  
                                                                                
  Dependency of   eslint [dev]                                                  
                                                                                
  Path            eslint > eslint-utils                                         
                                                                                
  More info       https://npmjs.com/advisories/1118                             
                                                                                

                                                                                
                                 Manual Review                                  
             Some vulnerabilities require your attention to resolve             
                                                                                
          Visit https://go.npm.me/audit-guide for additional guidance           
                                                                                
                                                                                
  Moderate        Memory Exposure                                               
                                                                                
  Package         bl                                                            
                                                                                
  Patched in      >=0.9.5 <1.0.0 || >=1.0.1                                     
                                                                                
  Dependency of   zigbee-shepherd-converters                                    
                                                                                
  Path            zigbee-shepherd-converters > zigbee-herdsman > dissolve > bl  
                                                                                
  More info       https://npmjs.com/advisories/596                              
                                                                                
found 3 vulnerabilities (2 moderate, 1 critical) in 874627 scanned packages
  run `npm audit fix` to fix 2 of them.
  1 vulnerability requires manual review. See the full report for details.

```
npm install zigbee-herdsman@0.6.1
rm -r node_modules/
npm install

```
npm audit
                                                                                
                       === npm audit security report ===                        
[... 22 lines cut ...]
                                                                                
  Package         bl                                                            
                                                                                
  Patched in      >=0.9.5 <1.0.0 || >=1.0.1                                     
                                                                                
  Dependency of   zigbee-shepherd-converters                                    
                                                                                
  Path            zigbee-shepherd-converters > zigbee-herdsman > dissolve > bl  
                                                                                
  More info       https://npmjs.com/advisories/596                              
                                                                                
found 2 vulnerabilities (1 moderate, 1 critical) in 874609 scanned packages
  run `npm audit fix` to fix 1 of them.
  1 vulnerability requires manual review. See the full report for details.

```
At least within the "npm-shrinkwrap.json" is after the first "npm audit fix" a reference to the new version. By today: 
```
},

    "zigbee-herdsman": {
      "version": "0.6.1",
```  

Regards

wossix, 2019-08-28:
By the way, I tested version 0.2.0 of zigbee-herdsman but that brings at least me back to the same issue.

Koenkk, 2019-08-28:
The security issues are solved in zigbee-herdsman 0.6.0, this will be integrated in the dev after 1.6 (see #1888)

Aberixis7306, 2019-08-28:
@wossix 
Thank you very much for your hint, a "npm install zigbee-herdsman@0.1.17" did the trick!!
zigbee2mqtt@1.5.1 is now up and running again, so I didn't need to repair all my >20 devices. 

Once more thx

Elliott32224, 2019-08-29:
I got past the herdsman issue with this fix, but have a new issue: _YAMLException: can not read a block mapping entry; a multiline key may not be an implicit key at line 13, column 3..._ ???

tmechen, 2019-08-30:
same problem here but with the docker image koenkk/zigbee2mqtt:arm32v6  
given it has been build 6 weeks ago (july 16, according to dockerhub) i think this might just not be fixed in there?

dudu2082, 2019-08-30:
Hi guys. I have also problem with this. I used "npm install zigbee-herdsman@0.1.17" but it didn't help:
": Started zigbee2mqtt.
sie 30 22:17:38 malina npm[1131]: > zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
sie 30 22:17:38 malina npm[1131]: > node index.js
^Xsie 30 22:17:39 malina npm[1131]: internal/modules/cjs/loader.js:638
sie 30 22:17:39 malina npm[1131]:     throw err;
sie 30 22:17:39 malina npm[1131]:     ^
sie 30 22:17:39 malina npm[1131]: Error: Cannot find module 'zigbee-herdsman/dist/ziee'
sie 30 22:17:39 malina npm[1131]:     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
sie 30 22:17:39 malina npm[1131]:     at Function.Module._load (internal/modules/cjs/loader.js:562:25)
sie 30 22:17:39 malina npm[1131]:     at Module.require (internal/modules/cjs/loader.js:692:17)
sie 30 22:17:39 malina npm[1131]:     at require (internal/modules/cjs/helpers.js:25:18)
sie 30 22:17:39 malina npm[1131]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/zapp/cie.js:6:14)
sie 30 22:17:39 malina npm[1131]:     at Module._compile (internal/modules/cjs/loader.js:778:30)
sie 30 22:17:39 malina npm[1131]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
sie 30 22:17:39 malina npm[1131]:     at Module.load (internal/modules/cjs/loader.js:653:32)
sie 30 22:17:39 malina npm[1131]:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
sie 30 22:17:39 malina npm[1131]:     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
sie 30 22:17:39 malina npm[1131]: npm ERR! code ELIFECYCLE
sie 30 22:17:39 malina npm[1131]: npm ERR! errno 1
sie 30 22:17:39 malina npm[1131]: npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`sie 30 22:17:39 malina npm[1131]: npm ERR! Exit status 1
sie 30 22:17:39 malina npm[1131]: npm ERR!
sie 30 22:17:39 malina npm[1131]: npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
sie 30 22:17:39 malina npm[1131]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
sie 30 22:17:39 malina npm[1131]: npm ERR! A complete log of this run can be found in:
sie 30 22:17:39 malina npm[1131]: npm ERR!     /home/pi/.npm/_logs/2019-08-30T20_17_39_954Z-debug.log
sie 30 22:17:39 malina systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE                                                            sie 30 22:17:39 malina systemd[1]: zigbee2mqtt.service: Unit entered failed state.
sie 30 22:17:39 malina systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.                                                                               sie 30 22:17:40 malina systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
sie 30 22:17:40 malina systemd[1]: Stopped zigbee2mqtt."
My npm version: 6.11.2
Node version : 10.16.3

Koenkk, 2019-08-31:
@tmechen did you map a custom zigbee shepherd converters?

tmechen, 2019-08-31:
i mount https://github.com/Koenkk/zigbee-shepherd-converters, to add support for new devices, according to your guide, see following docker-compose

>zigbee2mqtt:  
    image: koenkk/zigbee2mqtt:arm32v6  
    container_name: hassio_zigbee2mqtt
    restart: always
    networks:
      - hassio
    volumes:
      - /srv/zigbee2mqtt/data:/app/data
      - /srv/zigbee2mqtt/extensions/homeassistant.js:/app/lib/extension/homeassistant.js
      - /srv/zigbee2mqtt/extensions/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    logging: *default-logging

Koenkk, 2019-08-31:
What if you remove, it does it start?

iz3man, 2019-09-02:
> I have also problem with this. I used "npm install zigbee-herdsman@0.1.17" but it didn't help

Same here. Tried everything suggested in this thread but nothing did help.

Koenkk, 2019-09-03:
Can you help me how to reproduce this issue? I've tried with below but I don't get this error. Also what is the output of ls `node_modules/zigbee-herdsman/dist` ?

```js
cd /tmp
git clone https://github.com/Koenkk/zigbee2mqtt.git
cd zigbee2mqtt
npm install
npm start
```

iz3man, 2019-09-03:
```
pi@hassbian:~ $ cd /tmp/
pi@hassbian:/tmp $ git clone https://github.com/Koenkk/zigbee2mqtt.git
Cloning into 'zigbee2mqtt'...
remote: Enumerating objects: 248, done.
remote: Counting objects: 100% (248/248), done.
remote: Compressing objects: 100% (166/166), done.
remote: Total 6885 (delta 163), reused 158 (delta 81), pack-reused 6637
Receiving objects: 100% (6885/6885), 8.32 MiB | 1.04 MiB/s, done.
Resolving deltas: 100% (4761/4761), done.
pi@hassbian:/tmp $ cd zigbee2mqtt/
pi@hassbian:/tmp/zigbee2mqtt $ npm i

> @serialport/bindings@2.0.8 install /tmp/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.16.3 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/tmp/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/tmp/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'

> core-js-pure@3.1.4 postinstall /tmp/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/core-js-pure
> node scripts/postinstall || echo "ignore"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

added 1343 packages from 584 contributors and audited 874627 packages in 351.128s
found 3 vulnerabilities (2 moderate, 1 critical)
  run `npm audit fix` to fix them, or `npm audit` for details
pi@hassbian:/tmp/zigbee2mqtt $ npm start

> zigbee2mqtt@1.5.1 start /tmp/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 9/3/2019, 3:17:52 PM Logging to directory: '/tmp/zigbee2mqtt/data/log/2019-09-03.15-17-51'
  zigbee2mqtt:info 9/3/2019, 3:17:53 PM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
  zigbee2mqtt:info 9/3/2019, 3:17:53 PM Starting zigbee-shepherd
/tmp/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:45
                    debug('The serialport ' + self._sp.path + ' is opened.');
                                                       ^

TypeError: Cannot read property 'path' of null
    at Timeout.setTimeout [as _onTimeout] (/tmp/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:45:56)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-03T13_17_54_260Z-debug.log
```

```
pi@hassbian:/tmp/zigbee2mqtt $ ls node_modules/zigbee-herdsman/dist/
areq  cc-znp  dissolve-chunks  lib  unpi  zcl-id  zcl-packet  ziee  zive  zstack-constants
```

Well, this is a new error ....

Koenkk, 2019-09-03:
@iz3man and others, please contact me on telegram (koenkk), in this way we can find and solve the problem faster.

iz3man, 2019-09-03:
```
pi@hassbian:~ $ cd /tmp/zigbee2mqtt/
pi@hassbian:/tmp/zigbee2mqtt $ git rev-parse HEAD
ac3b924065c6f2b9dd7433a6f193d1924b34bb7d
pi@hassbian:/tmp/zigbee2mqtt $ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working tree clean
pi@hassbian:/tmp/zigbee2mqtt $ ls node_modules/zigbee-herdsman/dist/
areq  cc-znp  dissolve-chunks  lib  unpi  zcl-id  zcl-packet  ziee  zive  zstack-constants
```

Koenkk, 2019-09-04:
Together with @iz3man I found the cause of this problem.

If `git status` shows the following:

```
modified:   npm-shrinkwrap.json
modified:   package.json
```

the problem can be fixed by running the following commands:

```
git checkout npm-shrinkwrap.json
git checkout package.json
rm -rf node_modules && npm install
npm start
```

For those who ran into this issues, did you follow the update commands of https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#6-for-later-update-zigbee2mqtt-to-the-latest-version?

I will close this issue for know, if somebody else has the same issue but cannot fixed it by the above commands please contact me on telegram (koenkk).

closed 2019-09-04 by Koenkk

dudu2082, 2019-09-09:
https://github.com/Koenkk/zigbee2mqtt/issues/1892#issuecomment-527437549
Thanks! This fixed my problem and now I have again working zigbee2mqtt service.

dudu2082, 2019-09-10:
False alarm. Still it not working properly. 😔

Koenkk, 2019-09-11:
Still the same error?

Psychokiller1888, 2019-09-11:
Still the same here on a fresh install and not an update

Koenkk, 2019-09-11:
Please contact me on telegram (koenkk)

Koenkk, 2019-09-15:
@douglasjam this is probably fixed in the dev branch (hassio edge), can you check?

drsnuggles1982, 2019-09-23:
Have the same issue after new installation.

npm install zigbee-herdsman@0.1.17

didn't helped

Anything else we can do?

Koenkk, 2019-09-23:
@drsnuggles1982 https://github.com/Koenkk/zigbee2mqtt/issues/1892#issuecomment-527837064

Radiotechniman, 2019-09-25:
Just tried the "solution" mentioned above but the issue remained
went trough the source code of "zigbee-herdsman" and noticed the folder "zigbee-herdsman/dist/deprecated/" was removed with v6.x 
even-though it is still being referenced in "Zigbee2mqtt/lib/zapp/cie.js" 

`const Ziee = require('zigbee-herdsman/dist/deprecated/ziee');
const Zive = require('zigbee-herdsman/dist/deprecated/zive');`

manually updated the package.json and set "zigbee-herdsman": "0.5.7" insted of the 7 that was mentioned, did a `npm install`  and that sorted the issue.

AxHa2111, 2019-09-27:
> Just tried the "solution" mentioned above but the issue remained
> went trough the source code of "zigbee-herdsman" and noticed the folder "zigbee-herdsman/dist/deprecated/" was removed with v6.x [...]

Good morning Radiotechniman, first let me say "thank you". I think this is the solution. I try it, but I think I do anything wrong. I found the cie.js and found the mistake(s) in the file. I found also the /opt/zigbee2mqtt/package.json file and I change the value from  "zigbee-herdsman": "^0.7.0" to     "zigbee-herdsman": "^0.5.7". But after I did npm install, I had the same problem, the old value is set to "^0.7.0". Sorry, but I am a Linux beginner :-(

eyeson, 2019-10-09:
I am still having this on a fresh install following the installation guide. None of the above posts seem to fix my issues either

njordan77, 2019-10-09:
hm, i'm completely new to z2m but initially ran on a blank RPI (stretch) machine in the same errors listed above. Is there any fix available that works, or only downgrade will help? Regards

877dev, 2019-10-11:
I'm totally new to this and had this issue, I followed the instructions to the letter.
But when it had installed I ran npm audit fix as suggested.

Running the commands above fixed the issue and now it's working for me.. :)

KHOne23, 2019-10-15:
Hi, i am facing the same Problem with fresh install on my Raspberry pi with openhabian. 
I deinstalled nodejs and use nodejs for arm and now it works fine.

alwashe, 2019-10-29:
https://github.com/Koenkk/zigbee2mqtt/issues/1892#issuecomment-527837064

This fixed my problem after update.

DMike92, 2019-12-12:
Hi, there !
I'm really sorry for this newbie question, but:
Does Mosquito (or equiv) should be installed somewhere ?
If yes, can it be installed on the same raspberry ?