Koenkk/zigbee2mqtt #1893: Cannot edit xiaomi smoke detector docs

gericho, 2019-08-26:
I'm trying to propose a mod on the doc file for the honeywell mija xiaomi smoke sensor, since the procedure to pair isn't 100% correct, but II'm unable to find it into the linked file on the top, can someone help?

`https://www.zigbee2mqtt.io/devices/JTYJ-GD-01LM_BW.html`

dirstel, 2019-08-26:
Just read the first line and sport the Link:

"To contribute to this page, edit the following [file](https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docgen/device_page_notes.js)"

gericho, 2019-08-26:
Yes thank you, but as mentioned I already clicked there but it opens the wrong file I think, eg. searching “smoke” for smoke detector doesn’t show any hits

dirstel, 2019-08-26:
...and search for "JTYJ-GD-01LM" :D

closed 2019-08-27 by gericho