Koenkk/zigbee2mqtt #1895: Samsung Smart things motion sensors (2015 and 2016 Models)

jimmyeao, 2019-08-26:
# Bug Report 
## What happened
Smart things sensor doesn't update temperature / one sensor doesnt detect motion
## What did you expect to happen
Temperature is updated
## How to reproduce it (minimal and precise)
Device paired successfully, and reports temp on initial connect
## Debug Info
zigbee2mqtt version: 1.51
CC253X firmware version: CC2531_DEFAULT_20190608.zip

Im unsure how to add this after reading: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html - but its a bit over my head, could use some help getting these set up?

jonnyrider, 2019-09-21:
I'm having the same issue, did you ever manage to fix this?

jimmyeao, 2019-09-21:
Sadly, not yet

jonnyrider, 2019-09-22:
Is this the same for all your motion sensors? I can't get any of mine to
work at the moment.

Koenkk, 2019-09-22:
Can you check with the latest dev branch?

jimmyeao, 2019-09-22:
Dumb question time... How do I move to the dev branch? Is it a reinstall?

jimmyeao, 2019-09-22:
ok, so both are reporting motion, but only the 2015 model is updating temperature, on the 2016 model temp is not updating and inital read of the value has it much higher than the working one. It also has trouble joining:
```
9/22/2019, 4:58:10 PM - info: Device '0x24fd5b000109d007' joined
9/22/2019, 4:58:10 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x24fd5b000109d007"}}'
9/22/2019, 4:58:10 PM - info: Starting interview of '0x24fd5b000109d007'
9/22/2019, 4:58:10 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x24fd5b000109d007"}}'
9/22/2019, 4:58:12 PM - error: Failed to interview '0x24fd5b000109d007', device has not succesfully been paired
9/22/2019, 4:58:12 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x24fd5b000109d007"}}'
9/22/2019, 4:58:31 PM - error: Failed to interview '0x24fd5b000109d007', device has not succesfully been paired
``` 
This is the 2016 model I believe

Koenkk, 2019-09-22:
When starting zigbee2mqtt, do you see a succes or failed to configure for this device?

jonnyrider, 2019-09-23:
The Dev branch works fine for me. One of the sensors isn't working, but
I've discovered that the sensor is faulty, not the software!

jimmyeao, 2019-09-23:
no, I get this:
```
9/23/2019, 6:16:08 PM - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-09-23.18-16-07'
9/23/2019, 6:16:08 PM - info: Starting zigbee2mqtt version 1.6.0 (commit #9e94148)
9/23/2019, 6:16:08 PM - info: Starting zigbee-herdsman...
9/23/2019, 6:16:10 PM - info: zigbee-herdsman started
9/23/2019, 6:16:10 PM - info: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
9/23/2019, 6:16:10 PM - info: Currently 2 devices are joined:
9/23/2019, 6:16:10 PM - info: 0x24fd5b000109d007 (0x24fd5b000109d007): STS-IRM-250 - SmartThings Motion sensor (2016 model) (EndDevice)
9/23/2019, 6:16:10 PM - info: Kitchen Motion Sensor (0x000d6f0005760125): 3325-S - SmartThings Motion sensor (2015 model) (EndDevice)
9/23/2019, 6:16:10 PM - info: Zigbee: disabling joining new devices.
9/23/2019, 6:16:10 PM - info: Connecting to MQTT server at mqtt://localhost
9/23/2019, 6:16:10 PM - info: Connected to MQTT server
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'zigbee2mqtt/0x24fd5b000109d007', payload '{"occupancy":true,"tamper":false,"battery_low":false,"linkquality":92,"battery":"0.00","voltage":2400,"temperature":38.59}'
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'zigbee2mqtt/Kitchen Motion Sensor', payload '{"temperature":22.89,"linkquality":115,"occupancy":true,"tamper":false,"battery_low":false}'
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.6.0","commit":"9e94148","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'homeassistant/binary_sensor/0x24fd5b000109d007/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x24fd5b000109d007","json_attributes_topic":"zigbee2mqtt/0x24fd5b000109d007","name":"0x24fd5b000109d007_occupancy","unique_id":"0x24fd5b000109d007_occupancy_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x24fd5b000109d007"],"name":"0x24fd5b000109d007","sw_version":"Zigbee2mqtt 1.6.0","model":"Motion sensor (2016 model) (STS-IRM-250)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'homeassistant/sensor/0x24fd5b000109d007/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x24fd5b000109d007","json_attributes_topic":"zigbee2mqtt/0x24fd5b000109d007","name":"0x24fd5b000109d007_temperature","unique_id":"0x24fd5b000109d007_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x24fd5b000109d007"],"name":"0x24fd5b000109d007","sw_version":"Zigbee2mqtt 1.6.0","model":"Motion sensor (2016 model) (STS-IRM-250)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'homeassistant/sensor/0x24fd5b000109d007/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x24fd5b000109d007","json_attributes_topic":"zigbee2mqtt/0x24fd5b000109d007","name":"0x24fd5b000109d007_battery","unique_id":"0x24fd5b000109d007_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x24fd5b000109d007"],"name":"0x24fd5b000109d007","sw_version":"Zigbee2mqtt 1.6.0","model":"Motion sensor (2016 model) (STS-IRM-250)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'homeassistant/binary_sensor/0x24fd5b000109d007/battery_low/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.battery_low}}","device_class":"battery","state_topic":"zigbee2mqtt/0x24fd5b000109d007","json_attributes_topic":"zigbee2mqtt/0x24fd5b000109d007","name":"0x24fd5b000109d007_battery_low","unique_id":"0x24fd5b000109d007_battery_low_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x24fd5b000109d007"],"name":"0x24fd5b000109d007","sw_version":"Zigbee2mqtt 1.6.0","model":"Motion sensor (2016 model) (STS-IRM-250)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'homeassistant/sensor/0x24fd5b000109d007/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x24fd5b000109d007","json_attributes_topic":"zigbee2mqtt/0x24fd5b000109d007","name":"0x24fd5b000109d007_linkquality","unique_id":"0x24fd5b000109d007_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x24fd5b000109d007"],"name":"0x24fd5b000109d007","sw_version":"Zigbee2mqtt 1.6.0","model":"Motion sensor (2016 model) (STS-IRM-250)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
9/23/2019, 6:16:10 PM - info: MQTT publish: topic 'homeassistant/sensor/0x000d6f0005760125/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/Kitchen Motion Sensor","json_attributes_topic":"zigbee2mqtt/Kitch [... cut ...] itchen Motion Sensor","json_attributes_topic":"zigbee2mqtt/Kitchen Motion Sensor","name":"Kitchen Motion Sensor_linkquality","unique_id":"0x000d6f0005760125_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f0005760125"],"name":"Kitchen Motion Sensor","sw_version":"Zigbee2mqtt 1.6.0","model":"Motion sensor (2015 model) (3325-S)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'

```

For the 2016 model, it always reports the temperature as 38.59c

Koenkk, 2019-09-23:
@jimmyeao this doesn't occur immediately after starting zigbee2mqtt, but within the first +- 20 seconds. 

- Please provide more logging
- Can you also share your `database.db` ?

jimmyeao, 2019-09-25:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3651249/log.txt)
https://1drv.ms/u/s!AmXy90qDVFW-tvQpC92yXj1fIwbBVA?e=7lsSaI for the db

Thanks!

Koenkk, 2019-09-25:
@jimmyeao the issue is that the device interview fails somehow.

- Can you try to re-pair it once more and keep it awake when pairing? (by pressing buttons on it)
- If it still fails, could you provide the logging when running with `DEBUG=* npm start` while pairing?

gsmckenzie, 2019-10-14:
I have a similar issue. 

`Oct 14 17:42:12 Pi-Z2M-U npm[3090]:   zigbee2mqtt:info 10/14/2019, 5:42:12 PM MQTT publish: topic 'homeassistant/sensor/0x000d6f000bba4fd8/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x000d6f000bba4fd8","json_attributes_topic":"zigbee2mqtt/0x000d6f000bba4fd8","name":"0x000d6f000bba4fd8_temperature","unique_id":"0x000d6f000bba4fd8_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f000bba4fd8"],"name":"0x000d6f000bba4fd8","sw_version":"Zigbee2mqtt 1.6.0","model":"Motion sensor (2015 model) (3325-S)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Oct 14 17:42:12 Pi-Z2M-U npm[3090]:   zigbee2mqtt:info 10/14/2019, 5:42:12 PM MQTT publish: topic 'homeassistant/binary_sensor/0x000d6f000bba4fd8/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x000d6f000bba4fd8","json_attributes_topic":"zigbee2mqtt/0x000d6f000bba4fd8","name":"0x000d6f000bba4fd8_occupancy","unique_id":"0x000d6f000bba4fd8_occupancy_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f000bba4fd8"],"name":"0x000d6f000bba4fd8","sw_version":"Zigbee2mqtt 1.6.0","model":"Motion sensor (2015 model) (3325-S)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Oct 14 17:42:12 Pi-Z2M-U npm[3090]:   zigbee2mqtt:info 10/14/2019, 5:42:12 PM MQTT publish: topic 'homeassistant/sensor/0x000d6f000bba4fd8/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000d6f000bba4fd8","json_attributes_topic":"zigbee2mqtt/0x000d6f000bba4fd8","name":"0x000d6f000bba4fd8_linkquality","unique_id":"0x000d6f000bba4fd8_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x000d6f000bba4fd8"],"name":"0x000d6f000bba4fd8","sw_version":"Zigbee2mqtt 1.6.0","model":"Motion sensor (2015 model) (3325-S)","manufacturer":"SmartThings"},"availability_topic":"zigbee2mqtt/bridge/state"}'`

But the device will only state

zigbee2mqtt:info 10/14/2019, 5:42:50 PM MQTT publish: topic 'zigbee2mqtt/0x000d6f000bba4fd8', payload '{"temperature":23.18,"linkquality":63}'

Something is going wrong with the occupancy data.

Koenkk, 2019-10-14:
Can you try with the latest zigbee2mqtt dev branch and re-pair it?

gsmckenzie, 2019-10-14:
When re-pairing should you delete the old config data for that device?

gsmckenzie, 2019-10-14:
Amazing! works a treat now!

BTW I removed it from the config before re-pairing.

gsmckenzie, 2019-10-14:
Spoke too soon. It is reporting tamper properly, but occupancy is still not working.

`ct 14 19:00:57 Pi-Z2M-U npm[3440]: zigbee2mqtt:info  2019-10-14T18:00:57: Device '0x000d6f000bba4fd8' is supported, identified as: SmartThings Motion sensor (2015 model) (3325-S)
Oct 14 19:00:57 Pi-Z2M-U npm[3440]: zigbee2mqtt:info  2019-10-14T18:00:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6f000bba4fd8","model":"3325-S","vendor":"SmartThings","description":"Motion sensor (2015 model)","supported":true}}'
Oct 14 19:00:57 Pi-Z2M-U npm[3440]: zigbee2mqtt:info  2019-10-14T18:00:57: MQTT publish: topic 'zigbee2mqtt/0x000d6f000bba4fd8', payload '{"temperature":23.18,"linkquality":65,"occupancy":false,"tamper":false,"battery_low":false}'
Oct 14 19:00:59 Pi-Z2M-U npm[3440]: zigbee2mqtt:info  2019-10-14T18:00:59: MQTT publish: topic 'zigbee2mqtt/0x000d6f000bba4fd8', payload '{"temperature":23.18,"linkquality":52,"occupancy":false,"tamper":false,"battery_low":false}'

Koenkk, 2019-10-14:
Can you provide the debug log when receiving occupancy messages? (Set log_level to debug)

gsmckenzie, 2019-10-15:
It seems to be working now!

`Oct 15 08:14:14 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:14:14: Received Zigbee message from '0x000d6f000bba4fd8', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":38,"extendedstatus":0}' from endpoint 1 with groupID 0
Oct 15 08:14:14 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:14: MQTT publish: topic 'zigbee2mqtt/0x000d6f000bba4fd8', payload '{"temperature":17.4,"linkquality":105,"occupancy":true,"tamper":true,"battery_low":false}'
Oct 15 08:14:28 Pi-Z2M-U npm[4565]: zigbee2mqtt:warn  2019-10-15T07:14:28: Device '0x000d6f000bba4fd8' left the network
Oct 15 08:14:28 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x000d6f000bba4fd8"}}'
Oct 15 08:14:35 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:35: Device '0x000d6f000bba4fd8' joined
Oct 15 08:14:35 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000d6f000bba4fd8"}}'
Oct 15 08:14:35 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:35: Starting interview of '0x000d6f000bba4fd8'
Oct 15 08:14:35 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000d6f000bba4fd8"}}'
Oct 15 08:14:35 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:14:35: Device '0x000d6f000bba4fd8' announced itself
Oct 15 08:14:41 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:14:41: Received Zigbee message from '0x000d6f000bba4fd8', type 'readResponse', cluster 'genBasic', data '{"modelId":"3325-S","manufacturerName":"CentraLite","powerSource":3}' from endpoint 1 with groupID 0
Oct 15 08:14:44 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:14:44: Received Zigbee message from '0x000d6f000bba4fd8', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"stackVersion":2}' from endpoint 1 with groupID 0
Oct 15 08:14:47 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:14:47: Received Zigbee message from '0x000d6f000bba4fd8', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"����������������"}' from endpoint 1 with groupID 0
Oct 15 08:14:54 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:54: Successfully interviewed '0x000d6f000bba4fd8', device has successfully been paired
Oct 15 08:14:54 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:54: Device '0x000d6f000bba4fd8' is supported, identified as: SmartThings Motion sensor (2015 model) (3325-S)
Oct 15 08:14:54 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000d6f000bba4fd8","model":"3325-S","vendor":"SmartThings","description":"Motion sensor (2015 model)","supported":true}}'
Oct 15 08:14:54 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:54: Configuring '0x000d6f000bba4fd8'
Oct 15 08:14:54 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:14:54: Received Zigbee message from '0x000d6f000bba4fd8', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":34,"extendedstatus":0}' from endpoint 1 with groupID 0
Oct 15 08:14:54 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:54: MQTT publish: topic 'zigbee2mqtt/0x000d6f000bba4fd8', payload '{"temperature":17.4,"linkquality":99,"occupancy":true,"tamper":false,"battery_low":false}'
Oct 15 08:14:56 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:14:56: Received Zigbee message from '0x000d6f000bba4fd8', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":34,"extendedstatus":0}' from endpoint 1 with groupID 0
Oct 15 08:14:56 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:14:56: MQTT publish: topic 'zigbee2mqtt/0x000d6f000bba4fd8', payload '{"temperature":17.4,"linkquality":102,"occupancy":true,"tamper":false,"battery_low":false}'
Oct 15 08:15:06 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:15:06: Received Zigbee message from '0x000d6f000bba4fd8', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":1762}' from endpoint 1 with groupID 0
Oct 15 08:15:06 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:15:06: MQTT publish: topic 'zigbee2mqtt/0x000d6f000bba4fd8', payload '{"temperature":17.62,"linkquality":102,"occupancy":true,"tamper":false,"battery_low":false}'
Oct 15 08:15:10 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:15:10: Succesfully configured '0x000d6f000bba4fd8'
Oct 15 08:15:44 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:15:44: Received Zigbee message from '0x000d6f000bba4fd8', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":32,"extendedstatus":0}' from endpoint 1 with groupID 0
Oct 15 08:15:44 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:15:44: MQTT publish: topic 'zigbee2mqtt/0x000d6f000bba4fd8', payload '{"temperature":17.62,"linkquality":84,"occupancy":false,"tamper":false,"battery_low":false}'
Oct 15 08:15:47 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:15:47: Received Zigbee message from '0x000d6f000bba4fd8', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":34,"extendedstatus":0}' from e [... cut ...] false}'
Oct 15 08:16:04 Pi-Z2M-U npm[4565]: zigbee2mqtt:debug 2019-10-15T07:16:04: Received Zigbee message from '0x000d6f000bba4fd8', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":34,"extendedstatus":0}' from endpoint 1 with groupID 0
Oct 15 08:16:04 Pi-Z2M-U npm[4565]: zigbee2mqtt:info  2019-10-15T07:16:04: MQTT publish: topic 'zigbee2mqtt/0x000d6f000bba4fd8', payload '{"temperature":17.62,"linkquality":65,"occupancy":true,"tamper":false,"battery_low":false}'`

closed 2019-12-21 by stale[bot]