Koenkk/zigbee2mqtt #1897: Osram Garden Pole Mini after a few days

copystring, 2019-08-27:
Hi,

I have a network of about 55 devices. Some from Xiaomi, some from Osram.
Using zigbee2mqtt version 1.5.1 with the latest source_routing firmware (CC2531_SOURCE_ROUTING_20190619.zip).
The whole thing works mostly with minor issues.
However one thing really bugs me. It is the  OSRAM Smart+ mini gardenpole RGBW. It works just fine. On, off, colors. I use a plan to  switch the light on at night and off in the morning automatically every day. All good.
Only after a few days of running it appears offline and cannot be switched anymore.
So I turned on debug and tried to reproduce. As expected it took a few days for that to happen.
Turning the gardenpole physically fully off, waiting for a minute and turning it back on does not bring it back into the network. Restarting zigbee2mqtt does not bring it back either.
Trying to switch the light on/off results in this:
https://pastebin.com/TfrHjiEV

Rebooting the raspberry makes the issue go away though. After that it works fine for a few more days until the issue appears again.

zigbee2mqtt split the log with log_level debug into 5 files. Only the last 3 files remain. I guess it rotates logs if there are more than 3. The logs are nearly 10mb each so I filtered the relevant file with only the gardenpole.
https://pastebin.com/JJfyPLqL
Not sure how to provide full logs with 10mb each. Pastebin does not allow that.

I am a bit clueless about this issue since this only happens with this device. Sometimes it triggers after a few days without me or the smarthome system doing anthing automtically. Other times it happens when the light is switched automatically. For that I only have a normal log. Not a debug log. I can attach this if needed.

Kind regards.

Felix

Tom85649, 2019-09-03:
I got three Poles and one mini Pole. It's all a bit unstable with the Osram stuff. Sometimes I can switch them on/off or change color etc with no problem.
 And then just seconds later they don't react anymore. It's also my experience that rebooting the Pi sometimes helps. When close to the coordinator they work fine. But as soon as the connection is via routers the problems come up. But no clue yet what really causes the problems. 😫

james-fry, 2019-09-09:
This seems very similar to an issue I am having with ikea outlets:

https://github.com/Koenkk/zigbee2mqtt/issues/1931

could it be the same root cause?

closed 2019-11-15 by stale[bot]