Koenkk/zigbee2mqtt #1905: device_options config section is not described

milakov, 2019-08-29:
Hi!

`device_options` is mentioned at https://www.zigbee2mqtt.io/configuration/device_specific_configuration.html in "Changing device type specific defaults" section, but available options are not described in https://www.zigbee2mqtt.io/configuration/configuration.html.
This is needed to have https://github.com/danielwelch/hassio-zigbee2mqtt/issues/213 fixed.

Koenkk, 2019-08-31:
added, thanks!

closed 2019-08-31 by Koenkk

milakov, 2019-08-31:
I am not sure it is sufficiet... I feel like all possible options under `device_options` should be described for @danielwelch to process them from hass.io config

Koenkk, 2019-09-01:
The options differ per device (see device page). @danielwelch could this section be a bit more free format?

danielwelch, 2019-09-02:
As far as I know, can’t really be free format if we’re talking entering options via the UI config schema. A nice way around this for other similar situations has been allowing the user to point to a file (like devices.yaml), which can be more free form.

Koenkk, 2019-09-02:
I can provide all options, but not all options are applicable to all devices (e.g. `temperature_precision`), how can we deal with this?

danielwelch, 2019-09-03:
I guess I don’t really understand from the docs how this is used. Setting defaults for all devices that support the given options? So I’d think it’d be quite common for a field to be available to all devices. I’d think this is already handled by zigbee2mqtt?

Koenkk, 2019-09-03:
Yes, when e.g. setting

```
device_options:
  occupancy_timeout: 130
```

all devices that support `occupancy_timeout` will use this value EXCEPT when overridden again by the device itself, e.g.

``` 
devices:
  '0x00158d0001d82999':
    friendly_name: 'my_occupancy_sensor'
    retain: true
    occupancy_timeout: 120
```