Koenkk/zigbee2mqtt #1906: [Feature request] MQTT topic in the published message

wsw70, 2019-08-29:
### Introduction 

The current message published does not include the topic it was published on:

```
{
  "linkquality": 34,
  "battery": 100,
  "voltage": 3035,
  "click": "single"
}
```

It would be useful to add the `topic` (or `device`) the message was published to:

```
{
  "linkquality": 34,
  "battery": 100,
  "voltage": 3035,
  "click": "single",
  "topic": "zigbee2mqtt/0x00158d0002f2d2a2"
}
```

### Rationale for the request

I use a Home Assistant [MQTT Sensor](https://www.home-assistant.io/components/sensor.mqtt/) as a hub for incoming messages from various protocols (currently Zigbee (via `zigbee2mqtt`) and RF433 (via a Sonoff RF Bridge)). 

While the distinction between devices is possible for RF433 (though the ID part of the message they send), it is not possible for Zigbee ones (since the payload does not mention the device). The workaround is to creaet one sensor per device but this breaks some architectural logic in some cases.

jvdmeent, 2019-09-05:
Or even better:
Make messages configurable so we can decide which info will be send or not.

For example in device-config add timestamp and deviceid: 

From this:
{
  "linkquality": 34,
  "battery": 100,
  "voltage": 3035,
  "click": "single"
}

to this:
{
  "timestamp": 1567687713,
  "deviceid": "0x00158d0002f2d2a2"
  "linkquality": 34,
  "battery": 100,
  "voltage": 3035,
  "click": "single"
}

What i really want is a clear differance between "status" messages and "heartbeats".

carldebilly, 2019-09-05:
That should be an opt-in feature because it will fill up the HomeAssistant dabase with redundant information!

wsw70, 2019-09-05:
> Make messages configurable so we can decide which info will be send or not

Now that I think of, due to the asynchronous nature of the communication, a unique UUID for each message would be welcome too.

wsw70, 2019-09-05:
> That should be an opt-in feature because it will fill up the HomeAssistant dabase with redundant information!

This is why selecting what is being sent would be useful. I do not care, for instance, about the voltage level which is in the current message. Nor about the link quality. 

But for others that may be interesting.

mcpicoli, 2019-10-14:
I would go farther. It would be nice to be able to completely reconfigure the format of the messages. For example, to connect Zigbee2MQTT to Thingsboard via its Gateway API, I would need to separate the device identification from its data. For example:

(In Thingsboard)
Topic: ```/v1/gateway/telemetry```
Message:```
{
	"Device_Name":
	{
		"ts": 1567687713,
		"temperature": 27.4,
		"humidity": 56
	},
	"Other_Device":
	{
		"ts": 1567687723,
		"temperature": 26.4,
		"humidity": 63
	}
}
```

And so on. In this case, even if we had only a single device, the differences are many. For once, the device name does _not_ goes inside the topic.

wsw70, 2019-12-13:
Just to keep the request live: would there be hope to have this feature implemented or should I just close it?

Koenkk, 2019-12-14:
It looks like there is no interest for this feature besides you, therefore it has a very low priority for me (and will probably never be completed).

wsw70, 2019-12-14:
OK, that's fair - thanks for the great software in any case, it is fantastic.

closed 2019-12-14 by wsw70