Koenkk/zigbee2mqtt #1909: Xiaomi RTCGQ11LM detecting motion when cleary no motion

florian-ka, 2019-08-30:
Is there a sensitivity setting for the Aqara Motion Sensors like it is for the Hue Motion Sensor? Because for me the Motions sensor from Aqara are triggering an occupancy 'true' even at night when there is no one walking around the sensor and people are sleeping
[image]

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: CC2531_SOURCE_ROUTING_20190619

hawkefly, 2019-09-03:
maybe jack and jerry

pcb1962, 2019-09-07:
I think you have a faulty sensor, I have 12 Aqara motion sensors around the house and I have never seen a false positive.

lolorc, 2019-09-09:
I've been wondering about false positives with some DJT11LM (vibration sensors)
Haven't really looked at my logs tho. I guess I should sniff my network for several days to see if the devices really send the info.

flok, 2019-11-08:
> 
> 
> I've been wondering about false positives with some DJT11LM (vibration sensors)
> Haven't really looked at my logs tho. I guess I should sniff my network for several days to see if the devices really send the info.

For me it was occupancy_timeout set to 0. The Motion sensors send a update every 55 Minutes and since it was set to 0 it send not false but true, these where my false positives.

closed 2020-01-15 by stale[bot]