Koenkk/zigbee2mqtt #1911: [documentation] Missing value of Xiaomi Aqara vibration sensor (DJT11LM)

Rotzbua, 2019-08-31:
# Bug Report

## What happened

Got more information than expected. Documentation states less values:
https://www.zigbee2mqtt.io/devices/DJT11LM.html

Additionally served angle***.
```json
{
"angle":138,"linkquality":5,"angle_x":6,"angle_y":2,"angle_z":-84,"angle_x_absolute":84,"angle_y_absolute":88,
"battery":100,"voltage":3095,
"action":"vibration"
}
```

## What did you expect to happen

Documented values.

## Conclusion

Not a real bug, but I wanted to write down a note. Thanks for your great project!

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: unknown, downloaded stable 07/2019

TLBM, 2019-08-31:
Hi!

If we are on this topic, the Aqara double key wired wall switch (Router) QBKG12LM	
https://www.zigbee2mqtt.io/devices/QBKG12LM.html also reports a temperature value, although it seems 5 °C too high, compared to an WSDCGQ11LM Xiaomi Aqara temperature, humidity and pressure sensor.

'{"power":0,"consumption":0.16,"temperature":33,"linkquality":26,"state_left":"OFF","state_right":"OFF"}

Koenkk, 2019-09-03:
It is known that not all values reported by a device are described in the documentation.

I think this should not be fixed for only this device, but for all devices. For this this information should be added directly in zigbee-shepherd-converters. To be continued.

closed 2019-11-09 by stale[bot]