Koenkk/zigbee2mqtt #1913: Add support Legrand Valena Next (7 418 11)

TRuHa, 2019-08-31:
I'm intent on adding a new Legrand device not currently supported.

http://www.legrand.es/valena-next/

I have a number of problems, I need advice since it is not clear to me in the documentation.

`zigbee2mqtt:warn 2019-8-31 6:17:45 PM `permit_join` set to  `true` in configuration.yaml.`
`zigbee2mqtt:warn 2019-8-31 6:17:45 PM Allowing new devices to join.`
`zigbee2mqtt:warn 2019-8-31 6:17:45 PM Set `permit_join` to `false` once you joined all devices.`
`zigbee2mqtt:info 2019-8-31 6:17:45 PM Zigbee: allowing new devices to join.`
`zigbee2mqtt:info 2019-8-31 6:17:45 PM zigbee-shepherd ready`
`zigbee2mqtt:info 2019-8-31 6:17:45 PM Connected to MQTT server`
`zigbee2mqtt:info 2019-8-31 6:17:45 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'`
`zigbee2mqtt:info 2019-8-31 6:17:45 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"ac3b924","coordinator":20190425,"log_level":"info","permit_join":true}'`
`zigbee2mqtt:info 2019-8-31 6:20:25 PM Successfully reenabled joining`
`zigbee2mqtt:warn 2019-8-31 6:20:39 PM Message without device!`
`zigbee2mqtt:warn 2019-8-31 6:20:45 PM Device with modelID ' Connected outlet' is not supported.`
`zigbee2mqtt:warn 2019-8-31 6:20:45 PM Please see:`
`https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`
`zigbee2mqtt:warn 2019-8-31 6:20:46 PM Device with modelID ' Connected outlet' is not supported.`
`zigbee2mqtt:warn 2019-8-31 6:20:46 PM Please see:`
`https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`
`zigbee2mqtt:warn 2019-8-31 6:20:54 PM Message without device!`
`zigbee2mqtt:info 2019-8-31 6:21:20 PM Device incoming...`
`zigbee2mqtt:info 2019-8-31 6:21:20 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'`

What is the next step to proceed?

Sorry for my English. I would appreciate a help.

Koenkk, 2019-09-01:
try with: 

```js
{
        zigbeeModel: [' Connected outlet'],
        model: 'todo',
        vendor: 'Legran',
        description: 'Valena next',
        supports: 'on/off',
        fromZigbee: [fz.state],
        toZigbee: [tz.on_off],
}
```

TRuHa, 2019-09-01:
Thank you very much for the help Koenkk, and little by little I understand how this works.

Unfortunately I still do not get pair the device.
I show you how I have it configured:

**/opt/zigbee2mqtt/device.js**

       {
                zigbeeModel: [' Connected outlet'],
                model: 'todo',
                vendor: 'Legrand',
                description: 'Base enchufe conectado',
                supports: 'Enchufe on/off',
                fromZigbee: [fz.state],
                toZigbee: [tz.on_off],
         }

**/opt/zigbee2mqtt/data/configuration.yaml**

        devices:
            '0x00047400000ad84f':
                 friendly_name: 'smartplug'
                 retain: false

**log file**

`zigbee2mqtt:info 2019-9-1 8:49:26 PM Starting zigbee-shepherd`
`zigbee2mqtt:info 2019-9-1 8:49:27 PM zigbee-shepherd started`
`zigbee2mqtt:info 2019-9-1 8:49:27 PM Coordinator firmware version: '20190425'`
`zigbee2mqtt:info 2019-9-1 8:49:27 PM Currently 1 devices are joined:`
`zigbee2mqtt:info 2019-9-1 8:49:27 PM smartplug (0x00047400000ad84f): unkown - undefined unknown (Router)`

I do not quite understand how the device identifies with the devices.js of zigbee-shepherd-converters.
In the generated document database.db shows this content:

> {"id":2,"type":"Router","ieeeAddr":"0x00047400000ad84f","nwkAddr":50831,"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","epList":[1],"zclVersion":2,"appVersion":0,"stackVersion":64,"hwVersion":1,"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,2820,64513],"outClusterList":[0,25,64513],"clusters":{}}},"dateCode":" \u000020181219\u0000\u0000\u0000\u0000\u0000","_id":"uYm1LQkdEFRoILcA"}{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}

Tell you that the first time you try to pair the device use the firmware Z-Stack_Home_1.2 but it did not get any results, it simply did nothing. I am currently using Z-Stack_3.0.x with the CC2531 device. I don't know if you may require any other configuration. The truth is that it seems that it fails to identify the zigbee device correctly and fails to associate it with the zigbee-shepherd-converters configuration.

Thank you very much again, to see if you can figure out how to solve and support these devices.

Koenkk, 2019-09-03:
Can you try with 

```js
   {
            zigbeeModel: [' Connected outlet\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
            model: 'todo',
            vendor: 'Legrand',
            description: 'Base enchufe conectado',
            supports: 'Enchufe on/off',
            fromZigbee: [fz.state],
            toZigbee: [tz.on_off],
     }
```

TRuHa, 2019-09-03:
The truth is that I have tried everything already.

With space before Connected, with the \u0000 and the truth is that I don't understand why it doesn't recognize the device from the list of devices.js

The matter that the device is correctly paired, has the green LED of correct connection.

Any ideas?

Always show the same message at startup.

`zigbee2mqtt:info 9/3/2019, 7:27:25 PM smartplug (0x00047400000ad84f): unkown - undefined unknown (Router)`

Koenkk, 2019-09-05:
I think you are modifying the wrong file, `/opt/zigbee2mqtt/device.js` is not correct, it should be `/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/devices.js`

TRuHa, 2019-09-05:
You're right ... I haven't noticed the error, that's why I didn't recognize the device.

Now I have the problem that does nothing when I send the order ON/OFF.
But that is another problem.

Thank you so much for everything @Koenkk .

closed 2019-09-05 by TRuHa