Koenkk/zigbee2mqtt #1914: start error

marktn, 2019-08-31:
Hello,

How to solve this error?

pi@raspberrypi:~ $ npm start
npm ERR! path /home/pi/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-08-31T17_55_55_204Z-debug.log

TLBM, 2019-08-31:
Hi!

Did you follow all the steps in
https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html ?
Especially 
```
cd /opt/zigbee2mqtt
npm start
```

A couple of days ago I just set it up from scratch again on my completly fresh Raspberry Pi 3B+ with Raspbian Buster lite and had no problems.

Could you please provide the output, if you start it with `cd /opt/zigbee2mqtt; DEBUG=* npm start`

Kind regards
Thomas

marktn, 2019-09-01:
Hi Thomas,

Thank for helping!
Yes i used these instructions, but no result.

The Output:

pi@raspberrypi:~ $ cd /opt/zigbee2mqtt; DEBUG=* npm start

> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:173
  throw generateError(state, message);
  ^
YAMLException: unexpected end of the stream within a single quoted scalar at lin                                                                                        e 21, column 1:

    ^
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.j                                                                                        s:167:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1                                                                                        73:9)
    at readSingleQuotedScalar (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml                                                                                        /loader.js:586:3)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:                                                                                        1364:13)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loade                                                                                        r.js:1089:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:                                                                                        1359:12)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loade                                                                                        r.js:1089:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:                                                                                        1359:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js                                                                                        :1519:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.j                                                                                        s:1575:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional log                                                                                        ging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-01T07_52_55_958Z-debug.log

closed 2019-11-07 by stale[bot]